kerberos-io/machinery #95: IP Cam - url problem

menschmachine-de, 2017-08-15:
hi thx for kerberos.io,

after a few hours of trying to compile on ubuntu i get kerberos.io finally to work.

I guess i found an issue with ipcam urls

My adress is 
http://user:pass@192.168.xxx.xxx/cgi-bin/mjpg/video.cgi?channel=1&subtype=2

the log writes:
15/08/2017 22:58:32.319 ERROR [trivial] OpenCV : can't open url of ip camera

when i look into capture.xml

```
> <IPCamera>
>         <url type="text">http://user:pass@192.168.xxx.xxx/cgi-bin/mjpg/video.cgi?channel=1&**amp**;subtype=2</url>
>         <frameWidth type="number">640</frameWidth>
>         <frameHeight type="number">480</frameHeight>
>         <delay type="number">500</delay>
>         <angle type="number">0</angle>
>     </IPCamera>
```
Its added a "amp" after &.
When i edit the capture.xml kerberos.io web interface dosen't work anymore.

Best regards

cedricve, 2017-08-15:
Hey,

Nice find indeed. I confirm the issue. Well I think we need to replace it on the machinery (&amp; -> &). On the web this is converted because & is a reserved keyword. Correcting it in the config will make the web crash.

Thanks for reporting!

cedricve, 2017-08-17:
hey @menschmachine-de,

Hmm can you confirm the if you replace the &amp; to & that the machinery itself works? 

Cédric

Mmaaikel, 2017-12-06:
Hi,

I ran into the same issue today, but you can’t change it to & because the xml reader can’t parse it, only &amp; is valid. Even in CDATA it is not working. (I use it with RSTP://, not http).
Any idea what else I could change? The url works in RSTP viewer apps.

closed 2018-05-09 by cedricve