kerberos-io/machinery #99: [WEB] "no data available" for "Latest Activity"

loki-sft, 2017-10-10:
Hey, 
I have an issue with the "WEB"-Interface.
At the Dashboard everything is working fine - Live View, Overview, Hour, Weekday.
But when I switch to "Latest Activity" it shows "no data available" and for the heatmap it is just blank.
When I go to the calendar it says "oeps, no detection is found at..."

However there are lots of recorded files at /etc/opt/kerberosio/capture/.

In the configuration - motion - step 1  to set a region the frame is empty too.
The html-cooding looks like this:
`<img src="" id="hull" class="doSomethingOnChange" width="600" height="480" style="width:600px; height:480px; background-color:#ccc;">`

The src-tag is just empty.
I set the hole kerberosio directory to 777 but it was not helping.

cedricve, 2017-10-11:
Hey @loki-sft,

Can you try to clear the image directory? Could be due to a corrupt image.

loki-sft, 2017-10-11:
Thank you!
I cleared the folder. Now its working again.

closed 2017-10-11 by loki-sft