kerberos-io/machinery #100: Why mp4  video not being saved?

gianricod, 2017-10-11:
Hello 
machinery compiled and installed on ubuntu  .
IP camera (actually it is mjpgstreamer) configured with MJPEG stream. It works fine.
Video  output with default settings. 

I am not able to get video written on disk, only images are saved.

Do you know what i am doing wrong, or can you suggest any action to further investigate the issue?

Logs below:

11/10/2017 14:43:08.109 INFO  [business] Capture: checking health status of camera.
11/10/2017 14:43:13.112 INFO  [business] Capture: checking health status of camera.
11/10/2017 14:43:17.172 INFO  [business] HullExpositor: activity detected from (85,61) to (194,230)
11/10/2017 14:43:17.173 INFO  [business] Heuristic is valid; numberOfChanges: 742, Sequence duration: 1
11/10/2017 14:43:17.173 INFO  [business] IoVideo: firing
11/10/2017 14:43:18.115 INFO  [business] Capture: checking health status of camera.
11/10/2017 14:43:18.179 INFO  [business] IoVideo: start new recording 1507725797_6-173916_homecam_85-61-194-230_742_70.mp4
11/10/2017 14:43:18.185 INFO  [business] IoVideo (OpenCV): start writing images
11/10/2017 14:43:18.185 INFO  [business] IoVideo: locked write thread
11/10/2017 14:43:18.659 INFO  [business] Executing IO devices for 1 detection(s)
11/10/2017 14:43:19.189 INFO  [business] IoVideo: writing image
11/10/2017 14:43:19.189 INFO  [business] IoVideo: initializing capture thread
11/10/2017 14:43:19.189 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:19.522 INFO  [business] IoVideo: writing image
11/10/2017 14:43:19.856 INFO  [business] IoVideo: writing image
11/10/2017 14:43:20.191 INFO  [business] IoVideo: writing image
11/10/2017 14:43:20.198 INFO  [business] HullExpositor: activity detected from (114,56) to (210,230)
11/10/2017 14:43:20.199 INFO  [business] Heuristic is valid; numberOfChanges: 1393, Sequence duration: 1
11/10/2017 14:43:20.199 INFO  [business] IoVideo: firing
11/10/2017 14:43:20.524 INFO  [business] IoVideo: writing image
11/10/2017 14:43:20.858 INFO  [business] IoVideo: writing image
11/10/2017 14:43:20.870 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:21.192 INFO  [business] IoVideo: writing image
11/10/2017 14:43:21.526 INFO  [business] IoVideo: writing image
11/10/2017 14:43:21.860 INFO  [business] IoVideo: writing image
11/10/2017 14:43:21.877 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:22.194 INFO  [business] IoVideo: writing image
11/10/2017 14:43:22.211 INFO  [business] HullExpositor: activity detected from (108,69) to (211,201)
11/10/2017 14:43:22.211 INFO  [business] Heuristic is valid; numberOfChanges: 173, Sequence duration: 1
11/10/2017 14:43:22.211 INFO  [business] IoVideo: firing
11/10/2017 14:43:22.528 INFO  [business] IoVideo: writing image
11/10/2017 14:43:22.862 INFO  [business] IoVideo: writing image
11/10/2017 14:43:23.117 INFO  [business] Capture: checking health status of camera.
11/10/2017 14:43:23.196 INFO  [business] IoVideo: writing image
11/10/2017 14:43:23.389 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:23.530 INFO  [business] IoVideo: writing image
11/10/2017 14:43:23.865 INFO  [business] IoVideo: writing image
11/10/2017 14:43:24.171 INFO  [business] Executing IO devices for 2 detection(s)
11/10/2017 14:43:24.199 INFO  [business] IoVideo: writing image
11/10/2017 14:43:24.227 INFO  [business] HullExpositor: activity detected from (73,78) to (196,230)
11/10/2017 14:43:24.227 INFO  [business] Heuristic is valid; numberOfChanges: 736, Sequence duration: 1
11/10/2017 14:43:24.228 INFO  [business] IoVideo: firing
11/10/2017 14:43:24.532 INFO  [business] IoVideo: writing image
11/10/2017 14:43:24.867 INFO  [business] IoVideo: writing image
11/10/2017 14:43:24.901 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:25.201 INFO  [business] IoVideo: writing image
11/10/2017 14:43:25.535 INFO  [business] IoVideo: writing image
11/10/2017 14:43:25.869 INFO  [business] IoVideo: writing image
11/10/2017 14:43:25.909 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:26.203 INFO  [business] IoVideo: writing image
11/10/2017 14:43:26.242 INFO  [business] HullExpositor: activity detected from (7,109) to (187,230)
11/10/2017 14:43:26.243 INFO  [business] Heuristic is valid; numberOfChanges: 3438, Sequence duration: 1
11/10/2017 14:43:26.243 INFO  [business] IoVideo: firing
11/10/2017 14:43:26.537 INFO  [business] IoVideo: writing image
11/10/2017 14:43:26.871 INFO  [business] IoVideo: writing image
11/10/2017 14:43:27.205 INFO  [business] IoVideo: writing image
11/10/2017 14:43:27.422 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:27.539 INFO  [business] IoVideo: writing image
11/10/2017 14:43:27.873 INFO  [business] IoVideo: writing image
11/10/2017 14:43:28.121 INFO  [business] Capture: checking health status of camera.
11/10/2017 14:43:28.207 INFO  [business] IoVideo: writing image
11/10/2017 14:43:28.541 INFO  [business] IoVideo: writing image
11/10/2017 14:43:28.875 INFO  [business] IoVideo: writing image
11/10/2017 14:43:28.935 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:29.181 INFO  [business] Executing IO devices for 2 detection(s)
11/10/2017 14:43:29.209 INFO  [business] IoVideo: writing image
11/10/2017 14:43:29.543 INFO  [business] IoVideo: writing image
11/10/2017 14:43:29.877 INFO  [business] IoVideo: writing image
11/10/2017 14:43:29.941 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:30.211 INFO  [business] IoVideo: writing image
11/10/2017 14:43:30.545 INFO  [business] IoVideo: writing image
11/10/2017 14:43:30.879 INFO  [business] IoVideo: writing image
11/10/2017 14:43:31.213 INFO  [business] IoVideo: writing image
11/10/2017 14:43:31.451 INFO  [business] IoVideo: grabbing images
11/10/2017 14:43:31.547 INFO  [business] IoVideo: writing image
11/10/2017 14:43:31.881 INFO  [business] IoVideo: end writing images
11/10/2017 14:43:31.881 INFO  [business] IoVideo: remove videowriter
11/10/2017 14:43:31.882 INFO  [business] IoVideo: unlocking write thread
11/10/2017 14:43:32.967 INFO  [business] IoVideo: closing capture thread
11/10/2017 14:43:33.124 INFO  [business] Capture: checking health status of camera.
11/10/2017 14:43:38.126 INFO  [business] Capture: checking health status of camera.

gianricod, 2017-10-12:
Hi

Investigated further and it looks like an error opening the videowriter:

OpenCV: FFMPEG: tag 0x00000021/'!???' is not found (format 'mp4 / MP4 (MPEG-4 Part 14)')'

I have set codec back to  H264/X264, i still get an error but at least opencv/ffmpeg falls back to avc1 and i get a video recorded:

OpenCV: FFMPEG: tag 0x34363258/'X264' is not supported with codec id 28 and format 'mp4 / MP4 (MPEG-4 Part 14)'
OpenCV: FFMPEG: fallback to use tag 0x31637661/'avc1'

cedricve, 2017-10-12:
hey, make sure you have FFMPEG installed with x264 support.

gianricod, 2017-10-13:
Thank you Cedric
I have compiled ffmpeg with support for x264 , but i think there is still  some issue on  that  ubuntu that is messing up with my version of the compiled ffmpeg.

Anyway, it is fine like that . It is just a testing/dev machine and as far as  videos are produced i am ok.

closed 2017-10-18 by gianricod