kerberos-io/machinery #101: Upon reconfigure,  startIOThread does not run anymore

gianricod, 2017-10-18:
Hi Cedric

I am experimenting here (Machinery 2.3.1) and found that if i update the config (reconfigure) or set system off then on (which still i guess calls  reconfigure), the thread started by startIOThread stops running (or is locked in some way as the while loop inside checkDetectionsContinuously does not run).

Another funny thing is that, despite the IOThread  not running, if i have Video as IO it stills saves videos to disk.  But all other IO's are not executed .
Tests done with  with Video,TCPSocket    Video,MQTT    TCPSocket, MQTT   (i have an experimental MQTT implementation) .

I searched old issue and have not found anything. 

thanks
gianrico

cedricve, 2017-10-18:
Hi @gianricod,

-- The thread started by startIOThread stops running.

Weird, what camera are you using and on which OS? Also can you send me the log file, once you see the system is locked? 

- Another funny thing is that, despite the IOThread not running, if i have Video as IO it stills saves videos to disk.

Makes sense, this is because the IoVideo thread is executed in the main thread, using the fire() method. This is because with the video output, we want the system to record immediately.

cedricve, 2017-10-18:
UPDATE

I've been looking at the flow and I have following conclusions:
1. The [startIoThread]](https://github.com/kerberos-io/machinery/blob/master/src/kerberos/Kerberos.cpp#L26) method is only called once on booting the machinery.
2. When reconfiguring the machinery object is [destroyed](https://github.com/kerberos-io/machinery/blob/master/src/kerberos/Kerberos.cpp#L173-L176) and constructed.
3. While reconfiguring the IoThread is continuously executing, [nevertheless the machinery object exists or not](https://github.com/kerberos-io/machinery/blob/master/src/kerberos/Kerberos.cpp#L370).

My first opinion is that the thread might crash if the system is at **point 3**, while the machinery is destroyed **point 2**. If this is true, then this crazy. I've been running this project for 3 years and the issue never occurred, not even on my performance benchmarks. 

Is this an issue you can replicate by ease?

gianricod, 2017-10-18:
Hi Cedric
I will try to collect all the info and post it here as soon as i can.

cedricve, 2017-10-20:
@gianricod just curious do you have a stable MQTT implementation?

gianricod, 2017-10-20:
@cedricve  i have started working to it using libmosquittopp.
 
it is very basic at the moment. 
it sends a message to a topic upon each call of save(image,json). the message contains the json object.  very similar to the tcpsocket in that sense.

```
gianrico@vlin:~/machinery/build$ mosquitto_sub -h localhost -t "kios/mqtt" -v
kios/mqtt {"regionCoordinates":[97,41,220,208],"numberOfChanges":483,"timestamp":"1508514764","microseconds":"6-471402","token":16}
kios/mqtt {"regionCoordinates":[7,40,290,207],"numberOfChanges":1676,"timestamp":"1508514764","microseconds":"6-967244","token":16}
```
i see you have an enhancement open for that, we can discuss it there.

it needs to be polished a bit with more options/settings (and maybe use cases as well).

cedricve, 2017-10-22:
Thanks @gianricod moving discussion to https://github.com/kerberos-io/machinery/issues/89.

closed 2017-10-22 by cedricve