kerberos-io/machinery #102: How to enable continuous recording ?

brucerbwoods, 2017-10-18:
Please forgive my ignorance if this is something I have overlooked, I cannot however find any way in the kerberos web interface or machinery configuration files to simply record continuous footage, regardless of whether or not any image processing is taking place, or events are recorded.

Is this possible ? If not is this a feature that will be added in future ?

I have multiple instances of web/machinery running in docker, each for a separate IP camera.

cedricve, 2017-10-18:
hey @brucerbwoods,

Thanks for your question, well for the moment we don't have continuous recording. Maybe one day we'll add it but for now we don't do it because we believe that you should rely on other recording software (ffmpeg/vlc).

brucerbwoods, 2017-10-18:
When you say you believe that we should rely on other recording software (ffmpeg/vlc), do you mean to record a stream directly from the camera or the output of machinery ?

cedricve, 2017-10-18:
Yeah directly from the camera, it makes sense because as far I understand your requirements you want to record continuously without the need of doing motion detection.

brucerbwoods, 2017-10-18:
Thanks for your prompt response, I'll look into options for doing this, though I would suggest adding/enabling such a feature in the near future, for example, recording multiple weeks of footage in segments of a selectable length possibly with a ringbuffer, to a network share.

I realise this may not fit the use case envisioned for kerberos, but it would be fairly essential for any business or individual looking to use it for the security of property and premises, who would consider image processing/motion detection a bonus feature on top of this. 
Also, for a business running many services in docker, it is perfectly architected for resilience.

In any case, thankyou for your time and efforts in creating kerberos.

cedricve, 2017-10-18:
Makes sense @brucerbwoods, however main reason why I'm not working on such features is due to priorities. I'm having a hard time managing all the different parts (Web interface, Machinery, KiOS, installer, cloud, ...). I'm hoping I'll find some funding in near future so I can hire some people to scale Kerberos.io. For the moment this is still a one man team. I definitely remember your feedback, and will come back to you if we can start working on this feature. Thanks @brucerbwoods

actionTurtle, 2020-04-04:
Just following up if this ever got any traction? Thanks!!

JBFUK, 2021-01-05:
Agreed this is an essential feature for any surveillance system along with some form of storage/capacity management (for example consume a maximum of xGB storage with recordings before starting to delete/replace the oldest).

Having just spun up an instance of the open-source version I'm surprised to see it's not an option.  I'm also struggling to find any documentation on the 'advanced' features.