kerberos-io/machinery #107: Support of EasyCap devices

gorbatiy87, 2017-10-29:
Hi, I'm trying to use USB EasyCap device (USBTV007) with KiOS and Raspberry Pi 3, EasyCap is used to convert video from analog camera, but machinery can't capture video stream.
EasyCap is recognized as /dev/video0 and I'm capable to capture video with ffmpeg.

cedricve, 2017-11-01:
Can you send me the output of `lsusb`

gorbatiy87, 2017-11-01:
I'm not at home now, will send full output later,
but device id is 1b71:3002.

gorbatiy87, 2017-11-01:
> [root@kios-18fa2ab6 ~]# lsusb
> Bus 001 Device 002: ID 0424:9514
> Bus 001 Device 001: ID 1d6b:0002
> Bus 001 Device 003: ID 0424:ec00
> Bus 001 Device 004: ID 1b71:3002

gorbatiy87, 2018-09-05:
Hi, I've got some time for experimenting with EasyCap, I installed opencv for python and tried to grab some images using EasyCap.
I found out that EasyCap doesn't support resolution changing:
`vcap = cv2.VideoCapture()
vcap.set(cv2.CAP_PROP_FRAME_WIDTH, 480)
vcap.set(cv2.CAP_PROP_FRAME_HEIGHT, 640)`
both _set_ commands calls returns false
I tried to build build machinery with removed _set_ calls, and resolution set to 720x480 (matching easyCap resolution):
`    void USBCamera::setImageSize(int width, int height)
    {
        Capture::setImageSize(width, height);
        try
        {
            //m_camera->set(CV_CAP_PROP_FRAME_WIDTH, m_frameWidth);
            //m_camera->set(CV_CAP_PROP_FRAME_HEIGHT, m_frameHeight);
            m_camera->set(CV_CAP_PROP_FOURCC, CV_FOURCC(m_fourcc[0], m_fourcc[1], m_fourcc[2], m_fourcc[3]));
        }
        catch(cv::Exception & ex)
        {
            throw OpenCVException(ex.msg.c_str());
        }
}`
After that I could see picture trough the web interface in kerberos-io, I guess that _set_ calls was throwing exceptions.
 
Is it possible to make patch?

cedricve, 2018-09-05:
@gorbatiy87 great search, to be honest I think we should create a seperate capture device for this and maybe inherit from the USBCapture class. I think that would be the most clean way.

gorbatiy87, 2018-09-05:
Sounds nice, do you need my assistance?

cedricve, 2018-09-05:
Well I would say you can give it a try https://doc.kerberos.io/2.0/machinery/contribute#adding-a-new-class.. I'm here to help you if you have any questions.

gorbatiy87, 2018-09-05:
Thanks, nice guide I'll try and will contact you later.

KMiNT21, 2019-08-18:
Any success? 
The same problem with EasyCap - images only, no video in /etc/opt/kerberosio/h264.