kerberos-io/machinery #108: kerberosio says "Illegal instruction"

sabaatworld, 2017-11-01:
I have installed everything using the RPi guide on my RPi 0 W. The server works but says that machinery is not running. When I ssh into RPi and run kerberosio it prints "Illegal instruction" and exits.

Any idea what is wrong?

cedricve, 2017-11-01:
Hey, can you try to manually run the machinery by writing the "kerberosio" command?

cedricve, 2017-11-01:
Can you debug with *gdb*?

    gdb kerberosio

projetsdiy, 2017-11-14:
I've the same problem (RPi Zero). The error returned by gdb is 
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0xb6d38b10 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0

Hope it will useful

cedricve, 2017-11-15:
will be fixed with #109

closed 2017-11-15 by cedricve