kerberos-io/machinery #113: compiling machinery gives error

loki-sft, 2017-11-19:
Hello, 
compiling machinery on debian gives me following error
openssl 1.1.0f ist installed

```
/root/machinery/build/mosquitto/src/handle_connect.c:410:79: error: dereferencing pointer to incomplete type 'X509_NAME_ENTRY {aka struct X509_name_entry_st}'
context->username = mosquitto__strdup((char *)ASN1_STRING_data(name_entry->value));
                                                                         ^~

gianricod, 2017-11-23:
Hi @loki-sft 
you might need libssl-dev 

gianrico

loki-sft, 2017-11-23:
libssl-dev 1.1.0f is allready installed   -.-

gianricod, 2017-11-23:
it is weird.
i have found a post on a same error stating:

"On Debian stretch, installing libssl1.0-dev (and removing libssl-dev) with apt worked fine. ... "

loki-sft, 2017-11-23:
Thank you very much!
I just removed libssl-dev 1.1.0f and installed 1.0
Now everything is compiled without error.

closed 2017-11-23 by loki-sft