kerberos-io/machinery #115: Kerberos.io CC licence -- does ND make sense? Why not BY-NC-SA ?

gianricod, 2017-11-24:
Hi @cedricve 
I am wondering whether the ND makes sense on github. Forked repo with modifications can be considered Derivative Works, so basically one cannot have a forked repo with some features added to kerberos.io .
Just my thoughts, i am sure you are not suing anyone for that,  but i think that ND is too restrictive and might prevent more people to contribute. 

I would consider BY-NC-SA.

thanks
gianrico

cedricve, 2017-12-05:
Hmm good point @gianricod and makes sense, I'll update it to BY-NC-SA.

davidjb, 2018-03-20:
I noticed the licensing issue too -- a change in license would help facilitate the open-source nature of the code and the project a whole.

With regards to Creative Commons, they recommend _against_ using any CC license for software (see https://creativecommons.org/faq/#can-i-apply-a-creative-commons-license-to-software).  Creative Commons suggests licenses from the [Free Software Foundation](https://www.gnu.org/licenses/license-recommendations.html) or the [Open Source Initiative](https://opensource.org/licenses) instead as they feature sections relevant to distribution and use of source code.   A useful tool is GitHub's https://choosealicense.com/, which distils the [various FSF/OSI options further](https://choosealicense.com/licenses/).

As @gianricod points out, the code currently uses the CC-BY-ND licence, which isn't a good fit for software as Creative Commons explains.  In particular, the NoDerivatives (ND) part of the [current CC license](https://creativecommons.org/licenses/by-nc-nd/4.0/) means that:

> If you remix, transform, or build upon the material, you may not distribute the modified material.

So in short, anyone currently using your source code isn't permitted improve it and distribute modifications; in particular, this means that Pull Requests or _any_ other changes placed publicly on GitHub or elsewhere aren't permitted as this is considered 'distribution'.   

The [network of forks](https://github.com/kerberos-io/machinery/network) and [pull requests](https://github.com/kerberos-io/machinery/pulls?q=is%3Apr+is%3Aclosed) are active with people making contributions and modifications already, plus the README reads as though you're keen to encourage contributions.  A change in license would ensure these activities can continue without confusion.  Making the code available under an appropriate software license will also help with integration with other software and projects.

So, in short, using any Creative Commons licenses for software is against CC's recommendations. Keen to help discuss in more depth if you've got particular aims or requirements of your licensing.  Fwiw, it sounds as though the GPL most closely matches what you're aiming for with CC's By Attibution and ShareAlike clauses.

KrishnaPG, 2018-09-13:
A strange choice of license indeed, especially given that this project uses and depends on the works such as tinyxml, rapidjson, opencv which are fairly licensed (not to mention the Raspberry PI so on which are permissively open licensed). If the goal is others should contribute back the modifications, MPL (mozilla public license) is one of the good licenses to consider.

It is not clear under this  CC-NC-ND license if the goal is 
- to the prohibit sale of the software (product) or
- to prohibit the use of this software in commercial surveillance systems (service, not product)

tumik, 2019-05-23:
What's the status on this?
The kerberos.io documentation states 

> Kerberos.io is open source so everyone can customize the source code to its needs and share it with the community under the CC-NC-ND license model.

However that doesn't make sense as CC-NC-ND prohibits customizing and sharing the source code.

cedricve, 2019-12-09:
Version will be updated to https://creativecommons.org/licenses/by-nc-sa/4.0/

closed 2019-12-09 by cedricve

cedricve, 2020-01-01:
License changed to BY-NC-SA 4.0 in new version 2.8.0