kerberos-io/machinery #120: Set retention period

vpetersson, 2017-12-17:
Hey guys,

First, good job on this. One of the best software I've seen in this space.

One thing I noticed however is the lack of retention period. In most settings, you only want to keep images/videos around for X days. Did I overlook this feature or is there non?

I've worked around this in the past using a simple `find [...] -delete`, but it's not a very elegant solution.

cedricve, 2018-03-11:
Correct, but might be hard to implement a cross platform solution. What we do on KiOS is start removing images once the SD card is getting full. For other operating systems we expect the developer/user implement an retention strategy.

closed 2018-03-11 by cedricve

vpetersson, 2018-03-15:
@cedricve Fair point regarding cross-platform. *However*, the industry standard in my experience is to set it in time (e.g. 30 days). That does of course assume that you have 30 days worth storage on the system of coruse, but assuming you do, it does solve the problem.

vpetersson, 2018-03-15:
Just out of curiosity - where is this done in KiOS? Would be happy to copy the logic.

dtbaker, 2018-07-03:
Did anyone find a nice way to keep images for 30 days? Happy for it to be a cron job and a `find` script. But hoping someone has a more elegant solution. Thanks!

vpetersson, 2018-07-21:
@dtbaker I solved it by just having a cronjob like this:

```
30 1 * * * /usr/bin/find /path/to/files -type f -ctime +2 -delete
```