kerberos-io/machinery #121: machinery core dumps

stickystyle, 2017-12-20:
Machinery had been running for months without too many issues, today I pulled the latest image because I wanted the MQTT support and now I am getting core dumps right after startup. I tried removing the MQTT config from io.xml and the issue still persists.

Docker logs look like this...
```
2017-12-20 18:04:31,967 CRIT Set uid to user 0
2017-12-20 18:04:31,990 INFO RPC interface 'supervisor' initialized
2017-12-20 18:04:31,990 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2017-12-20 18:04:31,990 INFO supervisord started with pid 8
2017-12-20 18:04:32,992 INFO spawned: 'machinery' with pid 12
2017-12-20 18:04:34,026 INFO success: machinery entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2017-12-20 18:04:42,105 INFO exited: machinery (terminated by SIGABRT (core dumped); not expected)
2017-12-20 18:04:43,107 INFO spawned: 'machinery' with pid 32
2017-12-20 18:04:44,130 INFO success: machinery entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2017-12-20 18:04:52,131 INFO exited: machinery (terminated by SIGABRT (core dumped); not expected)
```

If I jump into the container and (after stopping the running one) start `/usr/bin/kerberosio` I get this...
```
20/12/2017 18:10:37.025 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
20/12/2017 18:10:37.025 INFO  [trivial] Parameters passed from commandline:
20/12/2017 18:10:37.027 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
20/12/2017 18:10:37.028 INFO  [trivial] Final configuration:
- algorithm = DifferentialCollins
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 500
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://XXXXXX:XXXXXX@192.168.10.108:554/videoMain
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.brightness = 50
- captures.RaspiCamera.contrast = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.RaspiCamera.framerate = 20
- captures.RaspiCamera.saturation = 0
- captures.RaspiCamera.sharpness = 0
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket =
- clouds.S3.folder =
- clouds.S3.privateKey =
- clouds.S3.publicKey =
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 5000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- configuration = /etc/opt/kerberosio/config/config.xml
- expositor = Hull
- expositors.Hull.region = 779,588|781,28|588,48|377,31|208,63|32,45|33,625|191,591|347,600|456,572|556,601|659,629
[... 291 lines cut ...]
            "IPAddress": "",
            "IPPrefixLen": 0,
            "IPv6Gateway": "",
            "MacAddress": "",
            "Networks": {
                "bridge": {
                    "IPAMConfig": {},
                    "Links": null,
                    "Aliases": null,
                    "NetworkID": "2f46aca2f896ca7bca2611a227e464ba1db66d8d51581405e4da5aa153be1400",
                    "EndpointID": "",
                    "Gateway": "",
                    "IPAddress": "",
                    "IPPrefixLen": 0,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "MacAddress": "",
                    "DriverOpts": null
                }
            }
        }
    }
]
```

gianricod, 2017-12-28:
Just a guess 
it could be the throttler config missing from your Video io.

https://github.com/kerberos-io/machinery/blob/master/config/io.xml

stickystyle, 2018-01-10:
Yep, that looks to have been the issue. After I added the throttler elements in the file it began working again, thank you!

closed 2018-01-10 by stickystyle