kerberos-io/machinery #122: machinery crashes linux kernel module

denis4net, 2017-12-25:
```
[37674.179084] ------------[ cut here ]------------
[37674.179107] WARNING: CPU: 2 PID: 26681 at fs/sysfs/group.c:237 sysfs_remove_group+0x5d/0x60
[37674.179115] sysfs group 'power' not found for kobject 'media0'
[37674.179119] Modules linked in: mousedev ir_xmp_decoder ir_mce_kbd_decoder ir_sharp_decoder ir_sanyo_decoder ir_sony_decoder ir_jvc_decoder ir_rc6_decoder ir_nec_decoder ir_rc5_decoder evdev uvcvideo videobuf2_vmalloc r8188eu(C) videobuf2_memops videobuf2_v4l2 ir_lirc_codec lirc_dev cfg80211 videobuf2_core sunxi_cir rfkill sun4i_codec sun8i_codec_analog snd_soc_core sun8i_ths snd_pcm_dmaengine snd_pcm cpufreq_dt uio_pdrv_genirq thermal_sys uio gpio_keys cuse fuse
[37674.179277] CPU: 2 PID: 26681 Comm: kerberosio Tainted: G        WC      4.11.12-sun8i #29
[37674.179283] Hardware name: Allwinner sun8i Family
[37674.179307] [<c010bad5>] (unwind_backtrace) from [<c01090df>] (show_stack+0xb/0xc)
[37674.179329] [<c01090df>] (show_stack) from [<c0463ecd>] (dump_stack+0x69/0x78)
[37674.179351] [<c0463ecd>] (dump_stack) from [<c0118295>] (__warn+0xa5/0xb8)
[37674.179370] [<c0118295>] (__warn) from [<c01182c9>] (warn_slowpath_fmt+0x21/0x2c)
[37674.179389] [<c01182c9>] (warn_slowpath_fmt) from [<c02361a9>] (sysfs_remove_group+0x5d/0x60)
[37674.179410] [<c02361a9>] (sysfs_remove_group) from [<c04d8999>] (device_del+0xa1/0x21c)
[37674.179433] [<c04d8999>] (device_del) from [<c05a3273>] (media_devnode_unregister+0x2b/0x38)
[37674.179457] [<c05a3273>] (media_devnode_unregister) from [<c05a2f2d>] (media_device_unregister+0xcd/0xd4)
[37674.179529] [<c05a2f2d>] (media_device_unregister) from [<bfa582ab>] (uvc_delete+0xd2/0xd8 [uvcvideo])
[37674.179588] [<bfa582ab>] (uvc_delete [uvcvideo]) from [<c04d7e31>] (device_release+0x1d/0x60)
[37674.179610] [<c04d7e31>] (device_release) from [<c0464d4d>] (kobject_put+0x45/0x90)
[37674.179631] [<c0464d4d>] (kobject_put) from [<c05a4175>] (v4l2_release+0x35/0x54)
[37674.179652] [<c05a4175>] (v4l2_release) from [<c01e4225>] (__fput+0x61/0x148)
[37674.179674] [<c01e4225>] (__fput) from [<c012b4db>] (task_work_run+0x8b/0xb4)
[37674.179696] [<c012b4db>] (task_work_run) from [<c011b133>] (do_exit+0x2c7/0x7f4)
[37674.179717] [<c011b133>] (do_exit) from [<c011b6a3>] (do_group_exit+0x27/0x84)
[37674.179739] [<c011b6a3>] (do_group_exit) from [<c0122067>] (get_signal+0x12b/0x484)
[37674.179762] [<c0122067>] (get_signal) from [<c01089df>] (do_signal+0x1c7/0x300)
[37674.179785] [<c01089df>] (do_signal) from [<c0108c3b>] (do_work_pending+0x63/0x70)
[37674.179806] [<c0108c3b>] (do_work_pending) from [<c0106081>] (slow_work_pending+0x9/0x18)
[37674.179815] ---[ end trace b3cd5ab0d5a7f5ec ]---
root@kios-83c5eeef:~# uname -an
Linux kios-83c5eeef 4.11.12-sun8i #29 SMP Mon Nov 20 00:25:54 CET 2017 armv7l armv7l armv7l GNU/Linux
```
```cat /etc/issue
Ubuntu 16.04.3 LTS \n \l

# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : Allwinner sun8i Family
Revision        : 0000
Serial          : 02c0008183c5eeef
```

I use kios rootfs in chroot for rpi3 on OrangePi PCas described [here](https://golos.io/ru--otkrytyij-kod/@denis4inet/kerberosio-na-orangepi-pc-or-sistema-videonablyudeniya-svoimi-rukami).

denis4net, 2017-12-25:
Looks like kernel issue: https://github.com/raspberrypi/linux/issues/1882

closed 2017-12-25 by denis4net

cedricve, 2017-12-25:
Thanks for reporting @denis4net

denis4net, 2017-12-25:
The main issue is webcamera device index changing after uvc driver exception.
So imagine, I setup `machinery` to use `/dev/video0`, it works for some time than uvc crashes and new camera device file is `/dev/video1`. Now I am trying to solve it using udev rules:
```
/etc/udev/rules.d# cat 80-webcam.rules
SUBSYSTEM=="usb", ATTR{idVendor}=="1908", ATTR{idProduct}=="2311", SYMLINK+="video10"
```

denis4net, 2017-12-25:
Could you add ability to set absolute path to v4l device, not just index? In that case it will be possible to use provided by the kernel symlinks like as:
```
root@kios-83c5eeef:/etc/udev/rules.d# ls -lsa /dev/v4l/by-id/
total 0
0 drwxr-xr-x 2 root root 60 Dec 25 07:43 .
0 drwxr-xr-x 4 root root 80 Dec 25 07:43 ..
0 lrwxrwxrwx 1 root root 12 Dec 25 07:43 usb-Generic_USB2.0_PC_CAMERA-video-index0 ->  ../../video0
```

cedricve, 2017-12-25:
@denis4net we use the OpenCV API to handle the v4l2 calls. Not sure if we can pass through abs path.

https://github.com/kerberos-io/machinery/blob/master/src/kerberos/capture/USBCamera.cpp#L132

denis4net, 2017-12-25:
Oh, yes, udev symlink doesn't help, because you use `cv::VideoCapture::open(int)`. OpenCV enumerates video devices itself.