kerberos-io/machinery #123: Throttler not working for multiple IO

poelie83, 2017-12-25:
When setting multiple IO commands (script, video & disk) the throttler for each specific IO action is not working as expected. For example, I set the disk throttler to 40 sec, the video to 10 and the script (that sends the video and message to my cellphone) to 60.

When motion is detected, I have more than 40 actions within one minute, also the script runs multiple times! even if the motion is there any more (6 to 10 minutes).

Can you help me out? when motion is detected I want to save a picture and record for x seconds en send the data to my mobile, its working but I am receiving multiple messages right now.

cedricve, 2018-02-01:
i'll look into this @poelie83 thanks

cedricve, 2018-02-01:
will be fixed in next release. https://github.com/kerberos-io/machinery/commit/5314cdf90b25415f3b4d154daec50111765300b2

closed 2018-02-01 by cedricve