kerberos-io/machinery #124: VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV

stinethebean3, 2017-12-31:
I'm getting this error `VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV` when starting kerberosio. 

here's my camera info: 
```
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : USB 2.0 Camera
	Bus info      : usb-3f980000.usb-1.3
	Driver version: 4.9.59
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'MJPG'
	Field             : None
	Bytes per Line    : 0
	Size Image        : 614989
	Colorspace        : Default
	Transfer Function : Default
	YCbCr/HSV Encoding: Default
	Quantization      : Default
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness (int)    : min=-128 max=127 step=1 default=-8193 value=0
                       contrast (int)    : min=0 max=255 step=1 default=57343 value=128
                     saturation (int)    : min=1 max=128 step=1 default=57343 value=64
                            hue (int)    : min=-40 max=40 step=1 default=-8193 value=0
 white_balance_temperature_auto (bool)   : default=1 value=1
                          gamma (int)    : min=72 max=500 step=1 default=57343 value=100
                           gain (int)    : min=0 max=100 step=1 default=57343 value=0
           power_line_frequency (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature (int)    : min=2800 max=6500 step=1 default=57343 value=4600 flags=inactive
                      sharpness (int)    : min=0 max=6 step=1 default=57343 value=3
         backlight_compensation (int)    : min=0 max=2 step=1 default=57343 value=1
                  exposure_auto (menu)   : min=0 max=3 default=0 value=3
              exposure_absolute (int)    : min=1 max=5000 step=1 default=625 value=625 flags=inactive
         exposure_auto_priority (bool)   : default=0 value=1
```
full log of error when starting kerberosio
```
Bad file [/tmp/logs/myeasylog.log]
Bad file [/tmp/logs/myeasylog.log]
Bad file [/tmp/logs/myeasylog.log]
Bad file [/tmp/logs/myeasylog.log]
31/12/2017 00:50:40.741 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
31/12/2017 00:50:40.742 INFO  [trivial] Parameters passed from commandline:
31/12/2017 00:50:40.744 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
31/12/2017 00:50:40.747 INFO  [trivial] Final configuration:
- algorithm = DifferentialCollins
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
[... 174 lines cut ...]
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 2
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Video
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.throttler = 0
- ios.Disk.timestampColor = white
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.GPIO.throttler = 0
- ios.MQTT.password = 
- ios.MQTT.port = 1883
- ios.MQTT.secure = false
- ios.MQTT.server = 1VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
Unable to stop the stream: Device or resource busy
```

stinethebean3, 2017-12-31:
This seems to have been a permissions issues. Leaving issue open as the error message was unhelpful.

cedricve, 2018-02-02:
thanks for sharing @stinethebean3, which permission issue? Starting kerberos with sudo helped?

closed 2018-03-11 by cedricve