kerberos-io/machinery #126: no timestamp on video

crottolo, 2018-01-08:
hello, after the tutorial installation on rpi with rpi camera noir v2.
At the end of the configuration encountered a problem with the timestamp on the video, 1280X720 30FPS on raspberry pi zero w.
dont'show the timestamp on captured video.

[image]
[image]

crottolo, 2018-01-08:
log file

[log.log](https://github.com/kerberos-io/machinery/files/1611586/log.log)

cedricve, 2018-01-19:
hey @crottolo,

Indeed we don't provide timestamping for the rapsberry pi camera, as we use the native library for recording (speeding up). However we can take a look to make it possible again. 

PS: if you would setup using the basic settings, you'll see a warning when trying to setup timestamping with the rpi camera module.

Cédric

crottolo, 2018-01-24:
thanks for support

crottolo, 2018-02-10:
For the future, is possible implement the timestamp?
I have 3 video cameras for video surveillance, when I brought the footage from the police said they prefer to have the timestamp, to validate the day and time.
This could also be useful for other users who use it for this purpose.
thanks

cedricve, 2018-02-10:
oh damn @crottolo,what happened? Yeah definitely makes sense!

crottolo, 2018-02-10:
Stolen brioches 😂😂
https://youtu.be/2aRF9k60Vts

cedricve, 2018-02-10:
is that filmed with @kerberosio? Which camera are you using.

crottolo, 2018-02-10:
This is the camera.. filmed Wurth Kerberos 

https://www.waveshare.com/rpi-ir-cut-camera.htm