kerberos-io/machinery #128: Add background timestamp

op3nus3r, 2018-01-13:
Hello everyone, thank you for this excellent application which I use with great pleasure on my Raspberry Pi 3. 
I use timestamping to mark captured images, during the day the image is white and the timestamping is written in black, everything is fine. 
During the night the image is black and the timestamp is always black, so you don't see it.
Is it possible to add a color background to the timestamp?

cedricve, 2018-01-14:
hey @op3nus3r, what I usually do is setting the timestamp color to red..

op3nus3r, 2018-01-14:
Hello @cedricve , thank you for your answer. 
That's what I also do, I added the red for the time stamp. I tried a little bit of every color. Under certain conditions, when the sun sets on a white wall or with yellow public lighting, the red timestamp is difficult to read as well.
It was just a suggestion to be able to configure a background color for timestamping if needed, it's not essential I thought I might not be the only one in this situation.

cedricve, 2018-01-14:
makes sense @op3nus3r, I think we'll add a background color by default then.. Let me assign a label to it, and add it to our next release.

op3nus3r, 2018-01-17:
Thank you very much for your answers, I don't doubt that this will be useful to a lot of people.

cedricve, 2018-02-01:
will be available in next release @op3nus3r, thanks for sharing!

commit 75217e43 (2018-02-01): draw date on a black background, fixes #128
closed 2018-02-01 by cedricve