kerberos-io/machinery #131: raspberry pi zero w and resolution 1920x1080

crottolo, 2018-01-16:
I have issue with raspberry pi zero w and resolution 1920x1080. 

Clean installation on raspbian strech.lite, update distro, update latest firmware, 128mb ram allocated.
the parameter for IOS.video.recordafter is 0 or 5 or 10 or 30, but the situation don't change.

this is the log.
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.codec = h264
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.directory = /etc/opt/kerberosio/capture/
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.enableHardwareEncoding = true
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.extension = mp4
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.fps = 10
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.markWithTimestamp = false
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.maxDuration = 30
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.privacy = false
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.recordAfter [OMX.broadcom.camera] SetParameter OMX_IndexParamPortDefinition failed : 80001005

crottolo, 2018-01-16:
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - configuration = /etc/opt/kerberosio/config/config.xml
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - expositor = Hull
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - expositors.Hull.region = 5,706|742,713|742,454|644,446|468,437|2,418
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - expositors.Rectangle.region.x1 = 0
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - expositors.Rectangle.region.x2 = 800
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - expositors.Rectangle.region.y1 = 0
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - expositors.Rectangle.region.y2 = 600
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristic = Sequence
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Counter.appearance = 3
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Counter.markers = 100,100|100,200|200,100|200,200
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Counter.maxDistance = 140
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Counter.minArea = 200
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Counter.minimumChanges = 20
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Counter.noMotionDelayTime = 100
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Counter.onlyTrueWhenCounted = true
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Sequence.minimumChanges = 20
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Sequence.minimumDuration = 0
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - heuristics.Sequence.noMotionDelayTime = 1000
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - io = Video
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Disk.directory = /etc/opt/kerberosio/capture/
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Disk.markWithTimestamp = true
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Disk.privacy = false
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Disk.throttler = 0
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Disk.timestampColor = white
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.GPIO.periodTime = 100000
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.GPIO.periods = 1
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.GPIO.pin = 17
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.GPIO.throttler = 0
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.MQTT.password =
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.MQTT.port = 1883
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.MQTT.secure = false
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.MQTT.server = 127.0.0.1
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.MQTT.throttler = 0
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.MQTT.topic = kios/mqtt
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.MQTT.username =
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.MQTT.verifycn = false
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Script.throttler = 0
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.TCPSocket.message = motion-detected
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.TCPSocket.port = 1337
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.TCPSocket.server = 127.0.0.1
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.TCPSocket.throttler = 0
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.codec = h264
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.directory = /etc/opt/kerberosio/capture/
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.enableHardwareEncoding = true
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.extension = mp4
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.fps = 10
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.markWithTimestamp = false
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.maxDuration = 30
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.privacy = false
Jan 17 00:16:17 raspberrypi kerberosio[4431]: - ios.Video.recordAfter [OMX.broadcom.camera] SetParameter OMX_IndexParamPortDefinition failed : 80001005
^C

cedricve, 2018-01-19:
Thanks.

**I have issue with raspberry pi zero w and resolution 1920x1080.**
What is the issue, you can set it?

**Clean installation on raspbian strech.lite, update distro, update latest firmware, 128mb ram allocated the parameter for IOS.video.recordafter is 0 or 5 or 10 or 30, but the situation don't change.**

Do you get short or too long videos?

crottolo, 2018-01-19:
the configuration for the raspi-cam

[image]

this is the result for this setting.
[image]

thanks for support. 
Your work is very beautiful

crottolo, 2018-01-19:
this is the log: 
Jan 19 18:45:19 cam01 kerberosio[1560]: 19/01/2018 18:45:19.833 INFO  [trivial] Final configuration:
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithm = DifferentialCollins
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.BackgroundSubtraction.dilate = 7
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.BackgroundSubtraction.erode = 5
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.BackgroundSubtraction.history = 15
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.BackgroundSubtraction.nmixtures = 5
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.BackgroundSubtraction.ratio = 1
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.BackgroundSubtraction.shadows = false
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.BackgroundSubtraction.threshold = 10
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.DifferentialCollins.erode = 5
Jan 19 18:45:19 cam01 kerberosio[1560]: - algorithms.DifferentialCollins.threshold = 9
Jan 19 18:45:19 cam01 kerberosio[1560]: - capture = RaspiCamera
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.IPCamera.angle = 0
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.IPCamera.delay = 500
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.IPCamera.frameHeight = 480
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.IPCamera.frameWidth = 640
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.IPCamera.url = rtsp://admin:888888@192.168.0.13/tcp/av0_1
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.angle = 0
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.brightness = 50
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.contrast = 0
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.delay = 500
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.frameHeight = 1080
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.frameWidth = 1920
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.framerate = 15
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.saturation = 0
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.RaspiCamera.sharpness = 0
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.USBCamera.angle = 0
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.USBCamera.delay = 500
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.USBCamera.deviceNumber = 0
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.USBCamera.frameHeight = 480
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.USBCamera.frameWidth = 640
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.VideoCapture.angle = 0
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.VideoCapture.delay = 500
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.VideoCapture.frameHeight = 1080
Jan 19 18:45:19 cam01 kerberosio[1560]: - captures.VideoCapture.frameWidth = 1920
Jan 19 18:45:20 cam01 kerberosio[1560]: - captures.VideoCapture.path = /tmp/video
Jan 19 18:45:20 cam01 kerberosio[1560]: - cloud = S3
Jan 19 18:45:20 cam01 kerberosio[1560]: - clouds.S3.bucket =
Jan 19 18:45:20 cam01 kerberosio[1560]: - clouds.S3.folder =
Jan 19 18:45:20 cam01 kerberosio[1560]: - clouds.S3.privateKey =
Jan 19 18:45:20 cam01 kerberosio[1560]: - clouds.S3.publicKey =
Jan 19 18:45:20 cam01 kerberosio[1560]: - condition = Enabled
Jan 19 18:45:20 cam01 kerberosio[1560]: - conditions.Enabled.active = true
Jan 19 18:45:20 cam01 kerberosio[1560]: - conditions.Enabled.delay = 50
Jan 19 18:45:20 cam01 kerberosio[1560]: - conditions.Time.delay = 10000
Jan 19 18:45:20 cam01 kerberosio[1560]: - conditions.Time.times = 23:01,19:59-8:01,19:59-8:01,19:59-8:01,19:59-8:01,19:59-8:01,19:59-8:01,19:59
Jan 19 18:45:20 cam01 kerberosio[1560]: - configuration = /etc/opt/kerberosio/config/config.xml
Jan 19 18:45:20 cam01 kerberosio[1560]: - expositor = Hull
Jan 19 18:45:20 cam01 kerberosio[1560]: - expositors.Hull.region = 5,706|742,713|742,454|644,446|468,437|2,418
Jan 19 18:45:20 cam01 kerberosio[1560]: - expositors.Rectangle.region.x1 = 0
Jan 19 18:45:20 cam01 kerberosio[1560]: - expositors.Rectangle.region.x2 = 800
Jan 19 18:45:20 cam01 kerberosio[1560]: - expositors.Rectangle.region.y1 = 0
Jan 19 18:45:20 cam01 kerberosio[1560]: - expositors.Rectangle.region.y2 = 600
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristic = Sequence
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Counter.appearance = 3
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Counter.markers = 100,100|100,200|200,100|200,200
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Counter.maxDistance = 140
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Counter.minArea = 200
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Counter.minimumChanges = 20
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Counter.noMotionDelayTime = 100
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Counter.onlyTrueWhenCounted = true
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Sequence.minimumChanges = 20
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Sequence.minimumDuration = 0
Jan 19 18:45:20 cam01 kerberosio[1560]: - heuristics.Sequence.noMotionDelayTime = 1000
Jan 19 18:45:20 cam01 kerberosio[1560]: - io = Video
Jan 19 18:45:20 cam01 kerberosio[1560]: - ios.Disk.directory = /etc/opt/kerberosio/capture/
Jan 19 18:45:20 cam01 kerberosio[1560]: - ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
Jan 19 18:45:20 cam01 kerberosio[1560]: - ios.Disk.markWithTimestamp = true
[... cut ...]
Jan 19 18:45:35 cam01 kerberosio[1599]: - ios.Video.fps = 15
Jan 19 18:45:35 cam01 kerberosio[1599]: - ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
Jan 19 18:45:35 cam01 kerberosio[1599]: - ios.Video.markWithTimestamp = false
Jan 19 18:45:35 cam01 kerberosio[1599]: - ios.Video.maxDuration = 30
Jan 19 18:45:35 cam01 kerberosio[1599]: - ios.Video.privacy = false
Jan 19 18:45:35 cam01 kerberosio[1599]: - ios.Video.recordAfter[OMX.broadcom.camera] SetParameter OMX_IndexParamPortDefinition failed : 80001005
^C
pi@cam01:~ $

marcel31415, 2018-02-07:
This error occurs for all Raspberry Pis with the RaspiCam. One working resolution is 1280x960.
Everything above that causes the same picture / error.
If you switch back to 2.3.1. it works fine. Seems as if they changed the drivers.

cedricve, 2018-02-07:
correct, we changed libraries to support hardware encoding. I should make a complete list of all working resolutions. Your assumptions/tests are correct @marcel31415.

marcel31415, 2018-02-07:
A list would be great. Is there a higher working resolution than 1280x960?

crottolo, 2018-02-10:
thanks for support.
PIzero w work great, i have tried resolution HD (1280*720*30FPS) and work.
For the next release is possible include full HD?

lahdekorpi, 2019-04-08:
Just confirmed the same issue with my RasPi-B on v2.7.2.
Downgrading to v2.3.1 works.