kerberos-io/machinery #132: Error on fresh installation of KiOS for Pi3 vB (Broadcom.Camera.Module) and infinte loop in boot

8ternity, 2018-01-24:
I download a fresh installation IMG of Kios and install on the SD Card. Everything was booting normaly except the internal Broadcom Camera module are crashing and KiOS are rebooting infintly from the machinery section.

Is that a fix for that?

cedricve, 2018-01-29:
@8ternity, make sure you have a Raspberry Pi camera connected or change the device type through the settings page (USB/IP camera).

gregorredinger, 2018-02-03:
@cedricve Thanks a lot, i had the same issue, because i use an usb cam instead of a raspberry cam.  changing the value in settings from RasperryCamera to USBCamera fixes the problem.

cedricve, 2018-02-03:
correct @gregorredinger, thanks for sharing!

closed 2018-02-28 by cedricve

Bechrissed, 2019-02-19:
Where do I change this setting? I just downloaded the installer and i'm never asked for a RaspberryCamer or USBCamera. I also see no file to edit after flashing.