kerberos-io/machinery #134: To new armbian?

Siggemada, 2018-01-28:
@bananapi:~$ ldd `which kerberosio` | grep "not found" 
libavcodec.so.56 => not found 
libavformat.so.56 => not found 
libavutil.so.54 => not found 
libswscale.so.3 => not found

cedricve, 2018-01-29:
make sure you have ffmpeg installed @Siggemada

Siggemada, 2018-01-29:
Yes, it is.

@bananapi:~$ ffmpeg -version
ffmpeg version 3.2.10-1~deb9u1 Copyright (c) 2000-2018 the FFmpeg developers
built with gcc 6.3.0 (Debian 6.3.0-18) 20170516
configuration: --prefix=/usr --extra-version='1~deb9u1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
libavutil      55. 34.101 / 55. 34.101
libavcodec     57. 64.101 / 57. 64.101
libavformat    57. 56.101 / 57. 56.101
libavdevice    57.  1.100 / 57.  1.100
libavfilter     6. 65.100 /  6. 65.100
libavresample   3.  1.  0 /  3.  1.  0
libswscale      4.  2.100 /  4.  2.100
libswresample   2.  3.100 /  2.  3.100
libpostproc    54.  1.100 / 54.  1.100

Siggemada, 2018-01-29:
Ok, it did not build. Something with mosquitto:

Scanning dependencies of target mosquitto
[ 61%] Building C object src/CMakeFiles/mosquitto.dir/conf.c.o
[ 62%] Building C object src/CMakeFiles/mosquitto.dir/conf_includedir.c.o
[ 63%] Building C object src/CMakeFiles/mosquitto.dir/context.c.o
[ 64%] Building C object src/CMakeFiles/mosquitto.dir/database.c.o
[ 64%] Building C object src/CMakeFiles/mosquitto.dir/handle_connack.c.o
[ 65%] Building C object src/CMakeFiles/mosquitto.dir/handle_connect.c.o
/home/xxx/machinery/build/mosquitto/src/handle_connect.c: In function ‘handle__connect’:
/home/xxx/machinery/build/mosquitto/src/handle_connect.c:410:6: warning: ‘ASN1_STRING_data’ is deprecated [-Wdeprecated-declarations]
      context->username = mosquitto__strdup((char *)ASN1_STRING_data(name_entry->value));
      ^~~~~~~
In file included from /usr/include/openssl/bn.h:31:0,
                 from /usr/include/openssl/asn1.h:24,
                 from /usr/include/openssl/objects.h:916,
                 from /usr/include/openssl/evp.h:27,
                 from /usr/include/openssl/x509.h:23,
                 from /usr/include/openssl/ssl.h:50,
                 from /home/xxx/machinery/build/mosquitto/lib/mosquitto_internal.h:28,
                 from /home/xxx/machinery/build/mosquitto/src/mosquitto_broker_internal.h:44,
                 from /home/xxx/machinery/build/mosquitto/src/handle_connect.c:22:
/usr/include/openssl/asn1.h:553:1: note: declared here
 DEPRECATEDIN_1_1_0(unsigned char *ASN1_STRING_data(ASN1_STRING *x))
 ^
/home/xxx/machinery/build/mosquitto/src/handle_connect.c:410:79: error: dereferencing pointer to incomplete type ‘X509_NAME_ENTRY {aka struct X509_name_entry_st}’
      context->username = mosquitto__strdup((char *)ASN1_STRING_data(name_entry->value));
                                                                               ^~
src/CMakeFiles/mosquitto.dir/build.make:182: recipe for target 'src/CMakeFiles/mosquitto.dir/handle_connect.c.o' failed
make[5]: *** [src/CMakeFiles/mosquitto.dir/handle_connect.c.o] Error 1
CMakeFiles/Makefile2:482: recipe for target 'src/CMakeFiles/mosquitto.dir/all' failed
make[4]: *** [src/CMakeFiles/mosquitto.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/mosquitto.dir/build.make:111: recipe for target 'mosquitto-prefix/src/mosquitto-stamp/mosquitto-build' failed
make[2]: *** [mosquitto-prefix/src/mosquitto-stamp/mosquitto-build] Error 2
CMakeFiles/Makefile2:786: recipe for target 'CMakeFiles/mosquitto.dir/all' failed
make[1]: *** [CMakeFiles/mosquitto.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

cedricve, 2018-01-29:
can you check which OpenSSL version is installed? 

    [root@kios-garage ~]# find / -name *ssl*
    /usr/lib/libevent_openssl-2.0.so.5
    /usr/lib/libssl.so
    /usr/lib/libevent_openssl.so
    /usr/lib/libevent_openssl-2.0.so.5.1.10
    /usr/lib/python2.7/lib-dynload/_ssl.so
    /usr/lib/python2.7/ssl.pyc
    /usr/lib/libssl.so.1.0.2
    /etc/ssl
    /etc/ssl/openssl.cnf

Looks like a version mismatch (https://github.com/openssl/openssl/issues/2239)

cedricve, 2018-01-29:
Also can you try to build with latest mosquitto repo. 

You can change this line (https://github.com/kerberos-io/machinery/blob/master/cmake/External-Mosquitto.cmake#L5) to **GIT_TAG maste**, and tray again.

Siggemada, 2018-01-29:
1.1.0f i think?

@bananapi:~$ openssl version
OpenSSL 1.1.0f  25 May 2017

@bananapi:~$ sudo find / -name *openssl*
[sudo] password for xxx:
/home/xxx/machinery/build/mosquitto/test/ssl/openssl.cnf
/etc/ssl/openssl.cnf
/var/cache/apt/archives/libcurl4-openssl-dev_7.52.1-5+deb9u4_armhf.deb
/var/lib/dpkg/info/openssl.conffiles
/var/lib/dpkg/info/openssl.postinst
/var/lib/dpkg/info/openssl.md5sums
/var/lib/dpkg/info/libcurl4-openssl-dev:armhf.md5sums
/var/lib/dpkg/info/libcurl4-openssl-dev:armhf.list
/var/lib/dpkg/info/openssl.list
/usr/include/openssl
/usr/include/openssl/opensslv.h
/usr/include/arm-linux-gnueabihf/openssl
/usr/include/arm-linux-gnueabihf/openssl/opensslconf.h
/usr/bin/openssl
/usr/share/man/man1/openssl.1ssl.gz
/usr/share/bash-completion/completions/openssl
/usr/share/lintian/overrides/openssl
/usr/share/doc/openssl
/usr/share/doc/libcurl4-openssl-dev
/usr/lib/arm-linux-gnueabihf/pkgconfig/openssl.pc
/usr/lib/arm-linux-gnueabihf/apr-util-1/apr_crypto_openssl-1.so
/usr/lib/arm-linux-gnueabihf/apr-util-1/apr_crypto_openssl.so
/usr/lib/arm-linux-gnueabihf/openssl-1.0.2
/usr/lib/ssl/openssl.cnf

cedricve, 2018-01-29:
thanks @Siggemada, can you try with my latest comment (changing version) of the cmake file.

Siggemada, 2018-01-29:
Looks like the problem remains. It did not feel like it was downloading anything new? I´ll try reboot.

@bananapi:~/machinery/build$ cmake .. && make && make check && sudo make install
External project: Googletest
External project: OpenCV
External project: Restclient
External project: Mosquitto
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xxx/machinery/build
[  7%] Built target opencv
[ 15%] Built target restclient
[ 22%] Built target googletest
[ 23%] Performing build step for 'mosquitto'
[ 25%] Built target libmosquitto_obj
[ 25%] Built target libmosquitto
[ 51%] Built target libmosquitto_static
[ 52%] Built target mosquittopp_obj
[ 53%] Built target mosquittopp
[ 55%] Built target mosquittopp_static
[ 58%] Built target mosquitto_pub
[ 60%] Built target mosquitto_sub
[ 61%] Building C object src/CMakeFiles/mosquitto.dir/handle_connect.c.o
/home/xxx/machinery/build/mosquitto/src/handle_connect.c: In function ‘handle__connect’:
/home/xxx/machinery/build/mosquitto/src/handle_connect.c:410:6: warning: ‘ASN1_STRING_data’ is deprecated [-Wdeprecated-declarations]
      context->username = mosquitto__strdup((char *)ASN1_STRING_data(name_entry->value));
      ^~~~~~~
In file included from /usr/include/openssl/bn.h:31:0,
                 from /usr/include/openssl/asn1.h:24,
                 from /usr/include/openssl/objects.h:916,
                 from /usr/include/openssl/evp.h:27,
                 from /usr/include/openssl/x509.h:23,
                 from /usr/include/openssl/ssl.h:50,
                 from /home/xxx/machinery/build/mosquitto/lib/mosquitto_internal.h:28,
                 from /home/xxx/machinery/build/mosquitto/src/mosquitto_broker_internal.h:44,
                 from /home/xxx/machinery/build/mosquitto/src/handle_connect.c:22:
/usr/include/openssl/asn1.h:553:1: note: declared here
 DEPRECATEDIN_1_1_0(unsigned char *ASN1_STRING_data(ASN1_STRING *x))
 ^
/home/xxx/machinery/build/mosquitto/src/handle_connect.c:410:79: error: dereferencing pointer to incomplete type ‘X509_NAME_ENTRY {aka struct X509_name_entry_st}’
      context->username = mosquitto__strdup((char *)ASN1_STRING_data(name_entry->value));
                                                                               ^~
src/CMakeFiles/mosquitto.dir/build.make:182: recipe for target 'src/CMakeFiles/mosquitto.dir/handle_connect.c.o' failed
make[5]: *** [src/CMakeFiles/mosquitto.dir/handle_connect.c.o] Error 1
CMakeFiles/Makefile2:482: recipe for target 'src/CMakeFiles/mosquitto.dir/all' failed
make[4]: *** [src/CMakeFiles/mosquitto.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/mosquitto.dir/build.make:111: recipe for target 'mosquitto-prefix/src/mosquitto-stamp/mosquitto-build' failed
make[2]: *** [mosquitto-prefix/src/mosquitto-stamp/mosquitto-build] Error 2
CMakeFiles/Makefile2:786: recipe for target 'CMakeFiles/mosquitto.dir/all' failed
make[1]: *** [CMakeFiles/mosquitto.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

cedricve, 2018-01-29:
indeed, make sure you have removed the **build/** folder before recompiling. Rebooting will not help.

Siggemada, 2018-01-29:
To make sure I got it right:

@bananapi:~/machinery/cmake$ more External-Mosquitto.cmake
message("External project: Mosquitto")

ExternalProject_Add(mosquitto
  GIT_REPOSITORY ${git_protocol}://github.com/eclipse/mosquitto.git
  GIT_TAG 8025f5a29b78551e1d5e9ea13ae9dacabb6830da
  SOURCE_DIR mosquitto
  BINARY_DIR mosquitto-build
  UPDATE_COMMAND ""
  PATCH_COMMAND ""
  CMAKE_GENERATOR ${gen}
  INSTALL_COMMAND mkdir -p ../thirdparty/lib/ && cp lib/cpp/libmosquittopp.a ../thirdparty/lib/
  CMAKE_ARGS
    -DCMAKE_INSTALL_PREFIX:PATH=${CMAKE_BINARY_DIR}/thirdparty
    -DWITH_UUID=no
    -DWITH_WEBSOCKETS=no
    -DWITH_SRV=no
)

set(MOSQUITTO_INCLUDE_DIR ${CMAKE_BINARY_DIR}/mosquitto/lib/ ${CMAKE_BINARY_DIR}/mosquitto/lib/cpp/)
set(MOSQUITTO_LIBRARY_DIR  ${CMAKE_BINARY_DIR}/mosquitto-build/lib/cpp/)

set(MOSQUITTO_LIBRARIES  mosquittopp.a crypto ssl)

include_directories(${MOSQUITTO_INCLUDE_DIR})

cedricve, 2018-01-29:
looks good.

Siggemada, 2018-01-29:
It will take a while ......

Siggemada, 2018-01-29:
A different error now. I have also tried to add package mosquitto-dev and libmosquitto-dev.

In file included from /usr/include/openssl/bn.h:31:0,
                 from /usr/include/openssl/asn1.h:24,
                 from /usr/include/openssl/objects.h:916,
                 from /usr/include/openssl/evp.h:27,
                 from /usr/include/openssl/x509.h:23,
                 from /usr/include/openssl/x509v3.h:14,
                 from /home/xxx/machinery/build/mosquitto/lib/tls_mosq.c:29:
/usr/include/openssl/asn1.h:553:1: note: declared here
 DEPRECATEDIN_1_1_0(unsigned char *ASN1_STRING_data(ASN1_STRING *x))
 ^
[ 89%] Building C object src/CMakeFiles/mosquitto.dir/__/lib/util_mosq.c.o
[ 91%] Building C object src/CMakeFiles/mosquitto.dir/websockets.c.o
[ 92%] Building C object src/CMakeFiles/mosquitto.dir/__/lib/will_mosq.c.o
[ 94%] Building C object src/CMakeFiles/mosquitto.dir/bridge.c.o
[ 96%] Linking C executable mosquitto
[ 96%] Built target mosquitto
Scanning dependencies of target mosquitto_passwd
[ 98%] Building C object src/CMakeFiles/mosquitto_passwd.dir/mosquitto_passwd.c.o
[100%] Linking C executable mosquitto_passwd
[100%] Built target mosquitto_passwd
[ 28%] Performing install step for 'mosquitto'
cp: cannot stat 'lib/cpp/libmosquittopp.a': No such file or directory
CMakeFiles/mosquitto.dir/build.make:73: recipe for target 'mosquitto-prefix/src/mosquitto-stamp/mosquitto-install' failed
make[2]: *** [mosquitto-prefix/src/mosquitto-stamp/mosquitto-install] Error 1
CMakeFiles/Makefile2:786: recipe for target 'CMakeFiles/mosquitto.dir/all' failed
make[1]: *** [CMakeFiles/mosquitto.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

cedricve, 2018-01-29:
hmm looks like the static file is missing..

    cp: cannot stat 'lib/cpp/libmosquittopp.a': No such file or directory

I remember using the develop branch, because it was only integrated over there. Can you try a last time with **GIT_TAG develop**

Siggemada, 2018-01-29:
No luck;

Scanning dependencies of target mosquitto_passwd
[ 98%] Building C object src/CMakeFiles/mosquitto_passwd.dir/mosquitto_passwd.c.o
[100%] Linking C executable mosquitto_passwd
[100%] Built target mosquitto_passwd
[ 28%] Performing install step for 'mosquitto'
cp: cannot stat 'lib/cpp/libmosquittopp.a': No such file or directory
CMakeFiles/mosquitto.dir/build.make:73: recipe for target 'mosquitto-prefix/src/mosquitto-stamp/mosquitto-install' failed
make[2]: *** [mosquitto-prefix/src/mosquitto-stamp/mosquitto-install] Error 1
CMakeFiles/Makefile2:786: recipe for target 'CMakeFiles/mosquitto.dir/all' failed
make[1]: *** [CMakeFiles/mosquitto.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

Siggemada, 2018-01-29:
lib/cpp is a link:

@bananapi:/lib$ ls -la
total 136
drwxr-xr-x 18 root root   4096 Nov 24 21:05 .
drwxrwsr-x 23 root samba  4096 Jan 17 20:48 ..
drwxr-xr-x  3 root root  12288 Jan 17 18:18 arm-linux-gnueabihf
drwxr-xr-x  2 root root   4096 Nov 17 14:58 bridge-utils
drwxr-xr-x  2 root root   4096 Nov 17 14:56 console-setup
lrwxrwxrwx  1 root root     21 Nov 17 15:04 cpp -> /etc/alternatives/cpp
drwxr-xr-x  3 root root   4096 Nov 17 14:58 crda
drwxrwxr-x 15 root root   4096 Nov 24 21:05 firmware
drwxr-xr-x  2 root root   4096 Nov 17 14:57 hdparm
drwxr-xr-x  2 root root   4096 Nov 17 14:52 ifupdown
drwxr-xr-x  2 root root   4096 Nov 17 14:40 init
-rwxr-xr-x  1 root root  54268 May  8  2016 klibc-wj8bPF_vWmgfISY5PwQIdrRif5U.so
lrwxrwxrwx  1 root root     30 Jun 15  2017 ld-linux-armhf.so.3 -> arm-linux-gnueabihf/ld-2.24.so
drwxr-xr-x  3 root root   4096 Nov 17 14:40 lsb
drwxr-xr-x  2 root root   4096 Nov 17 14:53 modprobe.d
drwxr-xr-x  3 root root   4096 Nov 24 21:02 modules
drwxr-xr-x  2 root root   4096 Nov 17 14:59 resolvconf
drwxr-xr-x  2 root root   4096 Nov 17 14:57 startpar
drwxr-xr-x  8 root root   4096 Nov 17 14:53 systemd
drwxr-xr-x 15 root root   4096 Sep  7 17:05 terminfo
drwxr-xr-x  4 root root   4096 Nov 17 14:58 udev
-----------------------------------

 sudo find / -name cpp -type d
/home/xxx/machinery/build/mosquitto-build/lib/cpp
/home/xxx/machinery/build/opencv/modules/core/misc/java/src/cpp
/home/xxx/machinery/build/opencv/modules/features2d/misc/java/src/cpp
/home/xxx/machinery/build/opencv/modules/java/generator/src/cpp
/home/xxx/machinery/build/opencv/samples/cpp
/home/xxx/machinery/build/mosquitto/test/lib/cpp
/home/xxx/machinery/build/mosquitto/lib/cpp
/home/xxx/machinery/build/googletest/googlemock/scripts/generator/cpp
/usr/share/doc/cpp
-----------------------------------------
I guess it is this one?

@bananapi:~/machinery/build/mosquitto-build/lib/cpp$ ls -la
total 52
drwxr-xr-x 3 xxx xxx 4096 Jan 29 19:35 .
drwxr-xr-x 4 xxx xxx 4096 Jan 29 19:35 ..
drwxr-xr-x 3  xxx xxx  4096 Jan 29 19:34 CMakeFiles
-rw-r--r-- 1  xxx xxx  2854 Jan 29 19:34 cmake_install.cmake
lrwxrwxrwx 1 xxx xxx    19 Jan 29 19:35 libmosquittopp.so -> libmosquittopp.so.1
lrwxrwxrwx 1  xxx xxx    24 Jan 29 19:35 libmosquittopp.so.1 -> libmosquittopp.so.1.4.90
-rwxr-xr-x 1  xxx xxx 26948 Jan 29 19:35 libmosquittopp.so.1.4.90
-rw-r--r-- 1  xxx xxx  7408 Jan 29 19:34 Makefile

cedricve, 2018-01-29:
yes it is. Hmm looks like only the dynamic libraries are compiled. It should compile the static ones though.

Siggemada, 2018-01-29:
Can this have anything to do with it?

CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_WITH_STATIC_CRT

cedricve, 2018-02-01:
sorry for the delay @Siggemada, I'm looking into this and come back with an explanation.

cedricve, 2018-02-01:
opened an issue in the main repo: https://github.com/eclipse/mosquitto/issues/687.

cedricve, 2018-02-02:
hey @Siggemada, I've provided a fix in the develop branch.Can you verify if it works by checking out  the develop branch?

   git checkout develop

Siggemada, 2018-02-02:
No problem. This is an exciting project. Looking forward to trying it out!

I get further but still fail on mosquitto:
[ 77%] Linking CXX executable ../../bin/kerberosio
/usr/bin/ld: cannot find -lmosquitto
collect2: error: ld returned 1 exit status
src/CMakeFiles/kerberosio.dir/build.make:615: recipe for target '../bin/kerberosio' failed
make[2]: *** [../bin/kerberosio] Error 1
CMakeFiles/Makefile2:1186: recipe for target 'src/CMakeFiles/kerberosio.dir/all' failed
make[1]: *** [src/CMakeFiles/kerberosio.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

cedricve, 2018-02-02:
how long ago did you started the compile? I've added a few more fixes, can you pull again?

Siggemada, 2018-02-02:
Ok, pulled and compiling now:

remote: Counting objects: 8, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 8 (delta 3), reused 8 (delta 3), pack-reused 0
Unpacking objects: 100% (8/8), done.
From https://github.com/kerberos-io/machinery
   c56a932..a7ba7df  develop    -> origin/develop
Updating c56a932..a7ba7df
Fast-forward
 cmake/External-Mosquitto.cmake | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

cedricve, 2018-02-02:
hey @Siggemada any update?

Siggemada, 2018-02-02:
Success! Built and installed.

Will test soon.

cedricve, 2018-02-02:
nice! can you also run **make install**, and send me a **.deb** file?

Siggemada, 2018-02-02:
Sure, you mean with dpkg-deb --build?

Can you walk me thru it?

cedricve, 2018-02-03:
**make install** will create the deb file

Siggemada, 2018-02-03:
I did run sudo make install as part of the instruction. But I can not find a .deb file.

I ran it again:
@bananapi:~/machinery/build$ sudo make install
[sudo] password for xxx:
[  7%] Built target opencv
[ 15%] Built target restclient
[ 22%] Built target googletest
[ 29%] Built target mosquitto
[ 32%] Built target ENCODE_LIBRARY
[ 32%] Built target rerun
[ 37%] Built target TINYXML_LIBRARY
[ 39%] Built target EXECUTOR_LIBRARY
[ 44%] Built target FILEWATCHER_LIBRARY
[ 59%] Built target KERBEROS_CORE
[ 77%] Built target kerberosio
[100%] Built target kerberos-test
Install the project...
-- Install configuration: ""
-- Up-to-date: /etc/opt/kerberosio/config/algorithm.xml
-- Up-to-date: /etc/opt/kerberosio/config/capture.xml
-- Up-to-date: /etc/opt/kerberosio/config/cloud.xml
-- Up-to-date: /etc/opt/kerberosio/config/condition.xml
-- Installing: /etc/opt/kerberosio/config/config.xml
-- Up-to-date: /etc/opt/kerberosio/config/expositor.xml
-- Up-to-date: /etc/opt/kerberosio/config/heuristic.xml
-- Up-to-date: /etc/opt/kerberosio/config/io.xml
-- Up-to-date: /etc/opt/kerberosio/config/stream.xml
-- Up-to-date: /etc/opt/kerberosio/scripts/run.sh
-- Installing: /etc/opt/kerberosio/logs/log.stash
-- Up-to-date: /etc/opt/kerberosio/symbols
-- Up-to-date: /etc/opt/kerberosio/capture/
-- Up-to-date: /etc/opt/kerberosio/h264/
-- Up-to-date: /etc/systemd/system/kerberosio.service
-- Up-to-date: /usr/bin/kerberosio

@bananapi:~/machinery/build$ sudo find / -name *.deb | grep kerberos
/home/xxx/armbian-machinery-kerberosio-armhf-2.4.0.deb

but that is the one I downloaded.

Siggemada, 2018-02-03:
Success!

[kerberosio_2.5.0_armhf.deb.zip](https://github.com/kerberos-io/machinery/files/1691606/kerberosio_2.5.0_armhf.deb.zip)

@bananapi:~/machinery/build$ sudo cpack
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: Kerberos
CPack: - Install project: Kerberos
CPack: Create package
CPack: - package: /home/xxx/machinery/build/kerberosio_2.5.0_armhf.deb generated.

cedricve, 2018-02-03:
nice! did you also tested it (just curious)?

Siggemada, 2018-02-03:
Yes, thanks! Works! Will try it out. Is there a forum?

cedricve, 2018-02-03:
we have gitter but it's not so active atm.  https://gitter.im/kerberos-io/hades

cedricve, 2018-02-03:
Let me know if you think everything works properly, than I publish it together with our new 2.6.0 release. You can close this issue is think everything is solved properly. Thanks for your efforts @Siggemada

Siggemada, 2018-02-03:
You might want to add a change permission step after sudo git clone. I did chown -R on the directory.

closed 2018-02-03 by Siggemada

cedricve, 2018-02-06:
Hey @Siggemada I made a small notification, are you able to compile once more and send me a new .deb file? Thanks.

Siggemada, 2018-02-07:
Sure!

[kerberosio_2.5.0_armhf.deb.zip](https://github.com/kerberos-io/machinery/files/1704670/kerberosio_2.5.0_armhf.deb.zip)

cedricve, 2018-03-04:
Hey @Siggemada,

Looks like we had another issue, are you able to fetch latest repo again and create a deb?
Thanks a lot @Siggemada

Siggemada, 2018-03-04:
git checkout develop?

Compiling now...

Siggemada, 2018-03-05:
Ok, here goes...
[kerberosio_2.5.0_armhf.deb.zip](https://github.com/kerberos-io/machinery/files/1781978/kerberosio_2.5.0_armhf.deb.zip)