kerberos-io/machinery #135: IPCAM Trendnet TV-IP310PI not working

merlink01, 2018-02-13:
I'm trying for hours now.

I always get: No Stream is the Machine running on Dashbord.

Here is the Log:

`
13/02/2018 12:56:12.785 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
13/02/2018 12:56:12.785 INFO  [trivial] Parameters passed from commandline:
13/02/2018 12:56:12.788 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
13/02/2018 12:56:12.790 INFO  [trivial] Final configuration:
- algorithm = BackgroundSubtraction
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 500
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 704
- captures.IPCamera.url = rtsp://admin:XXXX@192.168.178.71/streaming/Channels/2?tcp
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.brightness = 50
- captures.RaspiCamera.contrast = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.RaspiCamera.framerate = 20
- captures.RaspiCamera.saturation = 0
- captures.RaspiCamera.sharpness = 0
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.fourcc = MJPG
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket = 
- clouds.S3.folder = 
- clouds.S3.privateKey = 
- clouds.S3.publicKey = 
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 5000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- configuration = /etc/opt/kerberosio/config/config.xml
- expositor = Hull
- expositors.Hull.region = 779,588|781,28|588,48|377,31|208,63|32,45|33,625|191,591|347,600|456,572|556,601|659,629
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Counter
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 2
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Video
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.throttler = 0
- ios.Disk.timestampColor = white
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.GPIO.throttler = 0
- ios.MQTT.password = 
- ios.MQTT.port = 1883
- ios.MQTT.secure = false
- ios.MQTT.server = 127.0.0.1
- ios.MQTT.throttler = 0
- ios.MQTT.topic = kios/mqtt
- ios.MQTT.username = 
- ios.MQTT.verifycn = false
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.Script.throttler = 0
- ios.TCPSocket.message = motion-detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.TCPSocket.throttler = 0
- ios.Video.codec = h264
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.enableHardwareEncoding = true
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 30
- ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 30
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.throttler = 0
- ios.Video.timestampColor = white
- ios.Webhook.throttler = 0
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = true
- name = frontdoor
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.password = 75
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- streams.Mjpg.username = 
- timezone = Europe-Brussels
13/02/2018 12:56:12.793 INFO  [trivial] Logging is set to verbose
13/02/2018 12:56:12.794 INFO  [trivial] Starting capture device: IPCamera
13/02/2018 12:56:12.794 INFO  [trivial] Capture: Trying to open IP camera.
13/02/2018 12:56:12.794 INFO  [trivial] Capture: (Warning) You can change the capture device with the configuration files.
13/02/2018 12:56:14.176 INFO  [trivial] Capture: Succesfully opened IP camera.

`

I tried VLC  with MJPEG and H264.
All is fine in both configurations.

Also tried with and without ?tcp.

Any Ideas?

cedricve, 2018-02-17:
Hmm weird, can you run Kerberos from the commandline and see where it crashes?

dirstel, 2019-08-13:
Did you try

Picture
http://<user>:<password>@<IP der Kamera>//Streaming/Channels/1/picture

Stream
http://<user>:<password>@<IP der Kamera>//Streaming/Channels/1