kerberos-io/machinery #136: Curated list of all camera's that work properly with Kerberos.io

cedricve, 2018-03-07:
Hey all,

I've started this thread because people always thinker if there camera will work properly with Kerberos.io, before the actually start buying hardware. My goal is to inform those people, and create some kind of "certification", so we can guarantee specific cameras work.

Please report your camera if it is missing from this list.

[x] Raspberry Pi Camera v1.3
[x] Raspberry Pi Camera v2

palitu, 2018-03-29:
yeah, this is something that should be in their wiki page / docs.  It is one of my unknowns, what hardware to get?

makama80, 2018-04-16:
Hello Cédric, I am using an el-cheapo Wanscam HW0026 WiFi IP camera for quite a few weeks now. To my knowledge it works fine, so it was time to report it here.. I also have a RPi now, but it works also fine with Armbian.

Kind regards,

Martijn

wioxjk, 2020-08-31:
Curious if Axis M3065-V works too

MKE1138, 2022-05-28:
I've got a C270 but Kerberos doesn't recognize it. Is there something that needs to be done to enable it?