kerberos-io/machinery #137: Motion Detection - Record N seconds before motion and stop N seconds after motion

n3fariox, 2018-03-17:
Motion detection currently only records N seconds after motion is detected. 

Would be amazing to be able to be able to include N seconds before the motion event occurred, and stop recording after N seconds of no motion, to catch the surrounding footage.

Orabig, 2018-03-17:
Lol. Hillarious.

"Back to the future..."

You do understand that it means that the device should know in advance what
will happen in the future right ?

cedricve, 2018-03-17:
hey @n3fariox  @Orabig, well one way to solve this is to continuously save a buffer of the n last images in memory. Once motion is detected your start recording video, and you prepend the buffer. 

@n3fariox I have been thinking about this but didn't found the time to implement something sufficient. I'll keep this thread open, and close it once I got an appropriate solution.

Orabig, 2018-03-18:
I'm afraid that keeping a buffer of N seconds of (undoubtely uncompressed)
images would need much more memory than the Raspberry has. :(
I'm rather pessimistic about the feasability of this feature

n3fariox, 2018-03-18:
@cedricve Thanks for the consideration.
@Orabig :
1. I have 4 Amcrest cameras and they all support up to 5 seconds of pre-event recording when using on-board motion detection, and I'm pretty sure they're more resource constrained than Raspberri Pi's. tinyCam Monitor on Android even supports up to 10s of past recording.
2. You support many more distribution options than Raspberri Pi's. I'm currently running this via docker on an 8-core AMD build with 32GB of RAM.
3. This is an optional feature. You could opt not to buffer the past to save the RAM.

Orabig, 2018-03-18:
Ok, I apologize about my lack of knowledge about point 2. You made the
point.
My bad

shrimpza, 2018-05-29:
This would be an infinitely valuable addition to the system for me, and would allow me to switch to Kerberos as my primary camera solution.

cedricve, 2018-05-29:
Indeed we'll definitely do it. I just need to find to implement it or need someone else's help.

bradleywehmeier, 2018-07-27:
Cross-linking with matching FeatHub suggestion: [Add option to record pre and post event frames](https://feathub.com/kerberos-io/machinery/+18)

knaksbrig, 2018-09-10:
+1 for this feature

I've been evaluating a number of linux (Docker) surveillance systems recently and I've been impressed with the simplicity and stability of Kerberos.

The ability to include x seconds of video before a motion event is crucial for me. If this were included in Kerberos (along with object detection / highlighting) then we would have a clear winner.

CoReYeDe, 2018-10-29:
+1 for that Feature :D

jruth85, 2018-12-08:
Another +1 for this feature. This would be awesome.