kerberos-io/machinery #141: Machinery infinitive Loop, IPCamera HJT not working

dortizesquivel, 2018-04-01:
Hi guys,

After setting up Kerberos Web + Machinery in Raspbian following the doc, I always get: **"No Stream is the Machine running on Dashbord."**

Looking to Machinery's log it seems everything is ok "**01/04/2018 11:06:01.061 INFO  [trivial] Capture: Succesfully opened IP camera."** , but the logs keep printing the same constantly, not sure if the machinery process get restarted. is that normal?

The URL for the camera is **"rtsp://admin:admin123@192.168.1.211/11"** I tried it with VLC and everything works fine. Please see attached the log. Is there any option to make the machinery's log more verbose?  Thank you

`01/04/2018 11:05:59.274 INFO  [trivial] Logging is set to verbose
01/04/2018 11:05:59.274 INFO  [trivial] Starting capture device: IPCamera
01/04/2018 11:05:59.275 INFO  [trivial] Capture: Trying to open IP camera.
01/04/2018 11:05:59.275 INFO  [trivial] Capture: (Warning) You can change the capture device with the configuration files.
01/04/2018 11:06:01.061 INFO  [trivial] Capture: Succesfully opened IP camera.
01/04/2018 11:06:06.062 INFO  [business] Capture: checking health status of camera.
01/04/2018 11:06:06.063 INFO  [trivial] Stream: Configured stream on port 8889 with quality: 75
01/04/2018 11:06:06.064 INFO  [trivial] Starting cloud service: S3
01/04/2018 11:06:09.285 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
01/04/2018 11:06:09.285 INFO  [trivial] Parameters passed from commandline:
01/04/2018 11:06:09.288 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
01/04/2018 11:06:09.290 INFO  [trivial] Final configuration:
- algorithm = BackgroundSubtraction
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 600
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://admin:admin123@192.168.1.211/11
- cloud = S3
- clouds.S3.bucket =
- clouds.S3.folder =
- clouds.S3.privateKey =
- clouds.S3.publicKey =
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 5000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- configuration = /etc/opt/kerberosio/config/config.xml
- expositor = Rectangle
- expositors.Hull.region = 779,588|781,28|588,48|377,31|208,63|32,45|33,625|191,591|347,600|456,572|556,601|659,629
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 2
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Disk
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.throttler = 0
- ios.Disk.timestampColor = white
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.GPIO.throttler = 0
- ios.MQTT.password =
- ios.MQTT.port = 1883
- ios.MQTT.secure = false
- ios.MQTT.server = 127.0.0.1
- ios.MQTT.throttler = 0
- ios.MQTT.topic = kios/mqtt
- ios.MQTT.username =
- ios.MQTT.verifycn = false
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.Script.throttler = 0
- ios.TCPSocket.message = motion-detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.TCPSocket.throttler = 0
- ios.Video.codec = h264
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.enableHardwareEncoding = true
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 30
- ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 30
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.throttler = 0
- ios.Video.timestampColor = white
- ios.Webhook.throttler = 0
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = true
- name = frontdoor
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 15
- streams.Mjpg.password =
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- streams.Mjpg.username =
- timezone = Europe-Madrid`

dortizesquivel, 2018-04-01:
Based on the bad results obtained in my Raspberry Pi 3, I decided to give it a try to the same camera with same URL configuration but this time using Docker and my Macbook and surprisingly it worked.

So it seems this issue only affect my Raspberry, the camera settings are ok.
Do you have any suggestion of how I could debug why it didn't work in Raspberry?

cedricve, 2018-04-01:
hey @dortizesquivel, can you try changing the live stream port in the advanced settings? It might be that 8889 is already taken on your Raspbian installation.

dortizesquivel, 2018-04-01:
Hi @cedricve thanks for the reply, unfortunately 8889 was not taken, after running "**netstat -an | grep 8889**"it didn't show any entry. 

I have changed to 8888 (just in case) and still doesn't work. Looking to netstat results it seems Machinery never manage to bind port 8889, could I disable the stream?