kerberos-io/machinery #144: No Stream, is machinery running?

cloudybear, 2018-04-18:
I'm running two docker containers, each with an instance of kerberos running. Seems like one instance constantly have issue where camera and motion detection will die giving the error "No stream, is machinery running?" 

I'm unsure if it's a network issue, usually, I'd have to power cycle the camera that's connected to the instance with the machinery issue to fix the problem.

cedricve, 2018-04-20:
hey @cloudybear, hmm are they the same IP cameras?

cloudybear, 2018-04-20:
Ya they're both dlink docs cameras. It probably is the camera fed dying.
When I checked on the stream URL  it was just a blank blank screen.

solarplex, 2018-04-26:
I just received a new Raspi Camera and it seems my Pi Zero can not see it either. "No Stream, is machinery running?". I have three Raspi Cameras and only one works. These are all camera v2.1

cedricve, 2018-05-02:
@solarplex, this is weird. I have a couple of them running as well. Are you sure the connectors are all ok?

solarplex, 2018-05-20:
@cedricve I have tried different images, two different Pi Zero Ws, two different pi zero connectors, three different Pi Cameras. 

One item that could be the issue is that my camera bracket on both Pi Zero Ws have broke off. I have wedged two pieces of paper in there with the camera to keep it in place but I still have not been able to get a signal from either Pi on Kios.

What else can I do?

cedricve, 2018-05-22:
Thanks for updating @solarplex, I'm happy you got it resolved.

closed 2018-05-22 by cedricve