kerberos-io/machinery #147: incorrect image saving

Freakapotamus, 2018-05-19:
Among my saved images and video files I get images saved with a name like this "1526698333_6-235987_frontdoor_regionCoordinates_numberOfChanges_703.jpg"  Where it has no assigned region Coordinates or Number of changes.  Theses images also don't seem to be from valid motion triggers either.

Here is my current configuration:

`19/05/2018 12:48:29.869 INFO  [trivial] Final configuration:
- algorithm = DifferentialCollins
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 10
- algorithms.DifferentialCollins.threshold = 16
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 500
- captures.IPCamera.frameHeight = 1080
- captures.IPCamera.frameWidth = 1920
- captures.IPCamera.url = rtsp://Login:Password@192.168.1.14/live1.sdp
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.brightness = 50
- captures.RaspiCamera.contrast = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.RaspiCamera.framerate = 20
- captures.RaspiCamera.saturation = 0
- captures.RaspiCamera.sharpness = 0
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.fourcc = MJPG
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket = 
- clouds.S3.folder = 
- clouds.S3.privateKey = 
- clouds.S3.publicKey = 
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 4000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- configuration = /etc/opt/kerberosio/config/config.xml
- expositor = Hull
- expositors.Hull.region = 1868,1038|1866,441|1395,205|1348,298|1516,432|1516,528|845,806|20,821|12,1065|1151,1055|1427,920|1574,1045
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 3
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Disk,Video
- ios.Disk.directory = /mnt/PIHDD/camera/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.throttler = 0
- ios.Disk.timestampColor = none
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.GPIO.throttler = 0
- ios.MQTT.password = 
- ios.MQTT.port = 1883
- ios.MQTT.secure = false
- ios.MQTT.server = 127.0.0.1
- ios.MQTT.throttler = 0
- ios.MQTT.topic = kios/mqtt
- ios.MQTT.username = 
- ios.MQTT.verifycn = false
- ios.Pushbullet.throttler = 10
- ios.Pushbullet.token = o.mC5LPVCvPCphtSsEgWZQpFM86w9ciWQ3
- ios.Pushbullet.url = https://api.pushbullet.com
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.Script.throttler = 0
- ios.TCPSocket.message = motion-detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.TCPSocket.throttler = 0
- ios.Video.codec = h264
- ios.Video.directory = /mnt/PIHDD/camera/
- ios.Video.enableHardwareEncoding = true
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 7
- ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 30
- ios.Video.privacy = false
- ios.Video.recordAfter = 10
- ios.Video.throttler = 0
- ios.Video.timestampColor = none
- ios.Webhook.throttler = 0
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = false
- name = frontdoor
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 15
- streams.Mjpg.password = 
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- streams.Mjpg.username = 
- timezone = Australia-Melbourne`

cedricve, 2018-05-19:
@Freakapotamus, I suspect that the machinery is restarting due to problems (please check your logs) with your IP camera. Are you using a WIFI IP cam?

Freakapotamus, 2018-05-19:
Your right it seems when these types of files get created I get the following error in my log

```
19/05/2018 12:48:33.032 ERROR [trivial] OpenCV : can't open url of ip camera
19/05/2018 12:48:36.032 INFO  [trivial] Parameters passed from commandline:
19/05/2018 12:48:36.033 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
19/05/2018 12:48:36.038 INFO  [trivial] Final configuration:
```

No the IP camera is wired.

cedricve, 2018-05-19:
Which IP camera are you using?

Freakapotamus, 2018-05-19:
I am using a D-LINK DCS-4703E

cedricve, 2018-05-22:
Do you have any other endpoints than this url: rtsp://Login:Password@192.168.1.14/live1.sdp.

Freakapotamus, 2018-05-22:
I have 2 end points 
url: rtsp://Login:Password@192.168.1.14/live1.sdp
url: rtsp://Login:Password@192.168.1.14/live2sdp.

cedricve, 2018-06-06:
Hey @Freakapotamus  any update on this?

Freakapotamus, 2018-06-06:
I think its still happening.  I just wrote a script that deletes them if they are there that runs every 15 mins