kerberos-io/machinery #149: "checking health status of camera" loop ON Raspberry Pi 3B

marcel31415, 2018-06-03:
I get an error message after some time of using the camera does not seem to work anymore.
There is no stream on Port 8889 (or the port you are using) nor on the dashboard.
On the dashboard I receive the message:
"No stream, is the machinery running"

The system log on the webinterface says:
"Capture: checking health status of camera."

I've had this problem yesterday and rebooted. It all worked for about 15hours and then this error occured again.
I will not reboot, maybe i can provide any useful information from the current situation.

[kerberosio-configuration_checking camera health.tar.gz](https://github.com/kerberos-io/machinery/files/2065798/kerberosio-configuration_checking.camera.health.tar.gz)

cedricve, 2018-06-03:
Thanks, do you often view the livestream? It looks that the machinery get locked. Can you restart and check if it reoccurs.

marcel31415, 2018-06-03:
You mean:
Reboot and NOT watching the live stream (at 8889 or the dashboard) to prevent locking through both?

cedricve, 2018-06-04:
@marcel31415, I had an idea. If the issue reoccurs, can you check if it still receives motion?

marcel31415, 2018-06-04:
@cedricve 
*g* I've tested this the last time (but I will do it once again if the error occurs the next time) and there were not pictures saved at this moment. 
I will recheck that and will take a look at the log and whether files are saved.

How can i check if the stream is running without watching the stream ^^
Of course i can check the latest files, but that requires motion.

cedricve, 2018-06-04:
Hmm ok. If you have time to make it break again, and see if it saves images/videos that's great.
I found something interesting, looks like the stream is hanging at the stop function, and then it restarts. Looks like it "kind of crashed", but it remains detecting motion. But would be great if we can verify.

    2018-06-02 17:39:30,520 VERBOSE-1 [default] Logging is set to verbose
    2018-06-02 17:39:30,520 VERBOSE-1 [default] Stopping streaming
    2018-06-02 17:39:32,175 VERBOSE-2 [default] Capture: checking health status of camera.
    2018-06-02 17:39:37,175 VERBOSE-2 [default] Capture: checking health status of camera.
    2018-06-02 17:39:42,175 VERBOSE-2 [default] Capture: checking health status of camera.
    2018-06-02 17:39:47,176 VERBOSE-2 [default] Capture: checking health status of camera.
    2018-06-02 17:39:52,176 VERBOSE-2 [default] Capture: checking health status of camera.
    2018-06-02 15:40:54,716 VERBOSE-1 [default] Logging is written to: /etc/opt/kerberosio/logs/log.stash
    2018-06-02 15:40:54,721 VERBOSE-1 [default] Parameters passed from commandline:
    - config = /data/machinery/config/config.xml
    2018-06-02 15:40:54,725 VERBOSE-1 [default] Reading configuration file: /data/machinery/config/config.xml
    2018-06-02 15:40:54,732 VERBOSE-1 [default] Final configuration:

marcel31415, 2018-06-04:
ok. I will not watch the stream until tomorrow.
Then I will try to break it. If the stream stops shortly after doing that your suggestion is correct. As soon as I recognise the "kind of crash" i will send you the logs. Do you want me to check the loggin on the configuration before?

cedricve, 2018-06-04:
ok sounds cool, well the most important thing if you notice that the stream stops working. We need to verify if it's just the streams stopped, or that also motion detection stopped working (it doesn't record any media anymore). qFrom the logs you've send me I can conclude the stream is hanging, but it keeps recording video and taking images.

cedricve, 2018-06-04:
Also, do you also have an additional Rasberry Pi camera ?

marcel31415, 2018-06-05:
Stream is still running, I will now try to break it.

marcel31415, 2018-06-05:
That's interesting. The stream does not work ( I really didn't watch it the last two days). 
"2018-06-05 06:26:43,157 VERBOSE-1 [default] IoDisk: saving image 1528172789_6-128026_xxxxx123_691-323-717-361_53_262.jpg"
So it worked 45 minutes ago when i left home. I will now check if it will detect any motion when anybody else leaves home. Will get back to you in 30 min, latest

marcel31415, 2018-06-05:
This was when i detected that the stream was not working:
[20180605 0713kerberosio-configuration.tar.gz](https://github.com/kerberos-io/machinery/files/2070963/20180605.0713kerberosio-configuration.tar.gz)

It is still not working but pictures have been saved => motion works!
[20180605 0736kerberosio-configuration.tar.gz](https://github.com/kerberos-io/machinery/files/2070962/20180605.0736kerberosio-configuration.tar.gz)

So you were correct!

cedricve, 2018-06-05:
Looks like the streaming thread is dying. I'll try to simulate. Can you revert back to 2.6.1, and verify it doesn't happens there. Just to know if might be caused due to some changes in the new version.

marcel31415, 2018-06-05:
Ok, do you need any information before flashing the card? If not i will flash 2.6.1 tomorrow and will be able to give you information the day after tomorrow ;-)

cedricve, 2018-06-05:
ok thank you @marcel31415, great plan!

cedricve, 2018-06-06:
@marcel31415 I was wondering can you also see the recordings in the liveview (after the livestream breaks)?

marcel31415, 2018-06-06:
Pictures and Video are not corrupt.

cedricve, 2018-06-06:
Ok, that's "good". I'm wondering if it actually happens on 2.6.1 also. The strange thing is that at one point the machinery tries to stop, but crashes. Afterwards the system tries to restarts, I might suspect something is wrong with the camera connector, but it's too early to conclude that.

marcel31415, 2018-06-06:
The connector you are talking about is physical oder digital?
I am really excited to find out how 2.6.1 will work ;-)

cedricve, 2018-06-06:
Yeah indeed the physical connector of your rpi to your rpi camera module. Btw did you noticed a slow web interface, we just received a new issue. https://github.com/kerberos-io/kios/issues/32

marcel31415, 2018-06-06:
ok, so if everything works fine with 2.6.1 we can exclude that the connector is the problem. 
Will get back to you

cedricve, 2018-06-06:
That's correct, meanwhile I've marked the 2.7.0 release as prerelease.

cedricve, 2018-06-06:
Meanwhile I'm reverting the logging library to rule out this new library, which might cause issues with thread-safety.

marcel31415, 2018-06-06:
Flashed 2.6.1
wpa_supplicant error!

marcel31415, 2018-06-06:
> should be an error, on your side. make sure you have entered the correct
> credentials.

Mh. Flashed it three times. Will Check it

um 18:29:

cedricve, 2018-06-06:
make sure you test on 3B and not 3B+, it's not supported in 2.6.1

marcel31415, 2018-06-06:
It is an 3B :-/

um 18:44:

marcel31415, 2018-06-06:
I reflashed it again with 2.6.1 and tried another WiFi, same error.
Then i flashed 2.6.0 same thing. 
Tried two different RasPis. Tried other WiFis. Used Linux and Windows to flash. 

Error message is: 
Starting wpa_supplicant: failed (brcmfmac)

the /boot/wireless.conf seems to be correct. To typos or anything like that. 

Can you confirm this error?

cedricve, 2018-06-06:
nope should be ok..  for both versions. I remember you were able to use 2.6.1 no?

marcel31415, 2018-06-06:
Not any more. 270 and 261 cause this wpa error.

um 21:52:

marcel31415, 2018-06-06:
I have now tested 
3 Raspberry Pis 3 B (NOT +)
3 SD Cards
Two different WiFis.

marcel31415, 2018-06-06:
Must check my WiFi Network. It behaves strange. Not even normal Pis can connect anymore although one I installed earlier (last month) still works. 
Sorry.

cedricve, 2018-06-07:
Created a new release without the new logging library, looks like it's more stable.

cedricve, 2018-06-07:
A new release is available https://github.com/kerberos-io/kios/releases 2.7.1

marcel31415, 2018-06-08:
no wpa_supplicant error right from the start.
Stream works (until now, will wait until the weekend and close this thread if stream ist stable)

cedricve, 2018-06-08:
ok so it was due to the wifi connection?

cedricve, 2018-06-09:
hey @marcel31415, any update? Thanks!

marcel31415, 2018-06-10:
System is online for 2 days, 9 hours, 2 minutes, 23 seconds

No errors. Stream works.

closed 2018-06-10 by marcel31415