kerberos-io/machinery #150: ERROR [trivial] OpenCV : can't open url of ip camera.  But ffprobe says the stream is OK.

ChieftainY2k, 2018-06-07:
I am trying to hook up the machinery to the RTSP stream with the following config:

```
root@f2a02c61477f:/# cat /etc/opt/kerberosio/config/capture.xml

    <IPCamera>
        <url type="text">rtsp://172.20.0.2:9000/stream</url>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">480</frameHeight>
        <delay type="number">500</delay>
        <angle type="number">0</angle>
    </IPCamera>

```

The ffprobe says the stream is accessible and OK:

```
root@f2a02c61477f:/# ffprobe rtsp://172.20.0.2:9000/stream

ffprobe version n3.1.11-5-gac1ddc6361 Copyright (c) 2007-2017 the FFmpeg developers
  built with gcc 6.3.0 (Debian 6.3.0-18+deb9u1) 20170516
  configuration: --target-os=linux --enable-nonfree --enable-libx264 --enable-gpl
  libavutil      55. 28.100 / 55. 28.100
  libavcodec     57. 48.101 / 57. 48.101
  libavformat    57. 41.100 / 57. 41.100
  libavdevice    57.  0.101 / 57.  0.101
  libavfilter     6. 47.100 /  6. 47.100
  libswscale      4.  1.100 /  4.  1.100
  libswresample   2.  1.100 /  2.  1.100
  libpostproc    54.  0.100 / 54.  0.100
Input #0, rtsp, from 'rtsp://172.20.0.2:9000/stream':
  Metadata:
    title           : Unnamed
    comment         : N/A
  Duration: N/A, start: 351.803211, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p, 640x480, 25 fps, 25 tbr, 90k tbn, 180k tbc
```

But... the machinery says that OpenCV cannot open the ip camera:
```
root@f2a02c61477f:/# tail -f /etc/opt/kerberosio/logs/log.stash
...
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://172.20.0.2:9000/stream
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.brightness = 50
- captures.RaspiCamera.contrast = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.RaspiCamera.framerate = 20
- captures.RaspiCamera.saturation = 0
- captures.RaspiCamera.sharpness = 0
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.fourcc = MJPG
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket = ezg
- clouds.S3.folder = zeg
- clouds.S3.privateKey = eg
- clouds.S3.publicKey = zeg
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 5000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- configuration = /etc/opt/kerberosio/config/config.xml
- expositor = Hull
- expositors.Hull.region = 779,588|781,28|588,48|377,31|208,63|32,45|33,625|191,591|347,600|456,572|556,601|659,629
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 2
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Disk,MQTT
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.throttler = 0
- ios.Disk.timestampColor = none
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.GPIO.throttler = 0
- ios.MQTT.password = Df0_9GBvGyNj
- ios.MQTT.port = 10187
- ios.MQTT.secure = false
- ios.MQTT.server = m23.cloudmqtt.com
- ios.MQTT.throttler = 10
- ios.MQTT.topic = kios/mqtt
- ios.MQTT.username = xndxsbqn
- ios.MQTT.verifycn = false
- ios.Pushbullet.throttler = 10
- ios.Pushbullet.token = o.mC5LPVCvPCphtSsEgWZQpFM86w9ciWQ3
- ios.Pushbullet.url = https://api.pushbullet.com
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.Script.throttler = 0
- ios.TCPSocket.message = motion-detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.TCPSocket.throttler = 0
- ios.Video.codec = h264
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.enableHardwareEncoding = true
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 30
- ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 30
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.throttler = 0
- ios.Video.timestampColor = none
- ios.Webhook.throttler = 0
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = false
- name = example1
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 20
- streams.Mjpg.password =
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- streams.Mjpg.username =
- timezone = Europe-Jersey
07/06/2018 18:14:56.159 INFO  [trivial] Logging is set to info
07/06/2018 18:14:56.161 INFO  [trivial] Capture: Stop capture device
07/06/2018 18:14:56.161 INFO  [trivial] Capture: Start capture device: IPCamera
07/06/2018 18:14:56.162 INFO  [trivial] Capture: Trying to open IP camera.
07/06/2018 18:14:56.162 INFO  [trivial] Capture: (Warning) You can change the capture device with the configuration files.
07/06/2018 18:14:56.428 ERROR [trivial] OpenCV : can't open url of ip camera
```

What am I doing wrong ?
Tested the RTSP stream stream with both ffprobe and VLC , it plays just fine.

The stream is created with the following tools:
`raspivid -n -w 640 -h 480 -b 400000 -fps 30 -vf -hf -t 0 -o - | cvlc -vvv stream:///dev/stdin --sout '#rtp{sdp=rtsp://:9000/stream}' :demux=h264`

cedricve, 2018-06-19:
hey @ChieftainY2k is this resolved?

rotragit, 2018-08-15:
I had the same problem and was due to the presence of ffmpeg ubuntu package (3.x.x) and the kios ffmpeg version (cloned from git) on the same container. I guess is a incompatibility between opencv  (cloned as external from kios git) and the ffmpeg version. I suggest to try remove all ffmpeg end opencv packages and recompile.

cedricve, 2018-08-15:
@rotragit interesting. How can I reproduce this or how did you verified this issue? Is this issue occuring in current docker image?

rotragit, 2018-08-15:
Yes, get the Dockerfiles I have posted for i386 and add

`apt install -y ffmpeg`
before the git clone of the machinery.

I tryed to use bionic packages of opencv and ffmpeg removing the subtrees cloned by github, but opencv include in some source files reference to relative path in the source code, so it's doesn't compile at all. I didn't looked at the source files anyway. However  if the ubuntu packages for libopencv and ffmpeg are installed before the compilation then compiling goes ok but you get the error reported in the opencv library.

My guess is that the verion of opencv you git clone when compiling is incompatible with latest version of ffmpeg.

I verified the issue removing the ffmpeg package installation from the Dockerfile and all goes right.