kerberos-io/machinery #151: Add support for client_id in MQTT

dimaj, 2018-06-12:
Current implementation of `IoMQTT` does not support users to provide a custom `client_id`. 

The way I have my mosquitto server setup is that it requires all clients to have a specific prefix for a client_id. This makes the setup a little more secure in my opinion. 
Based on what I've seen in the code, the only things are are configurable are credentials, connection encryption, server host/port and topic.

Based on what I've seen, I'm guessing prefix is configured via constructor for `mosquittopp` 
 * https://github.com/eclipse/mosquitto/blob/master/lib/cpp/mosquittopp.cpp#L168-L170
 * https://github.com/eclipse/mosquitto/blob/5e60136449d678948520085985d7e7f91f9e601c/lib/mosquitto.h#L197...L203

dimaj, 2018-06-13:
After reading code some more, mosquitto is reinitialized with a client ID set to camera name. It would still be awesome if there was a separate field for client id, which would default to camera name if field is left blank

cedricve, 2018-06-19:
hmm @dimaj, looks like i've closed your PR because it was still pointing to master. Can restart your PR with the correct branch?

dimaj, 2018-08-10:
my PRs have been merged. This issue could be closed.

closed 2018-08-10 by dimaj