kerberos-io/machinery #153: Issues

Wonderbox2000, 2018-06-17:
Hi,

Thanks for you software, it's really fun !
But I think I'm having some issues.

None of my settings in config.xml or capture.xml seem to be saved.
Each time I switch off recordings from the WEBIU, the settings go back to default and the machinery won't restart automatically.
Same happens when  I try to change the setting in the motion section of the WEB IU

Do you have any suggestion to solve this ?

cedricve, 2018-06-17:
hello,

which version are you using, also how did you installed kerberos.io

Wonderbox2000, 2018-06-17:
Web | 2.5.1 
-- | -- | --
Machinery | 2.6.1

On RPI 3 and wired USB Cam

ChieftainY2k, 2018-06-17:
Hi.
Had same issue in my "dockerized" version.
If there is ANY non-xml or damaged/malformed xml file in the /config folder then the ConfigXMLReader->getConfigs(...) method tries to parse it anyway throwing exception (which is silently suppressed/caught) even before any data is saved.

Hope that helps.

Wonderbox2000, 2018-06-17:
Yes, I think it did the trick.
I had two backup files config.xml.bak and capture.xml.bak (I made them just in case...).
I feel like it's working now.

I have another question, can I trigger the Off/On button from a script ?
I want to trigger the recording when I press an Amazon Dash Button (I know how to make the Dash Button trigger a script, but I am looking for the command to trigger)

Thanks guys

closed 2018-06-17 by Wonderbox2000

cedricve, 2018-06-18:
Thanks @ChieftainY2k, indeed. Maybe we should document is somewhere.