kerberos-io/machinery #154: Trigger the On/Off button from a script

Wonderbox2000, 2018-06-17:
can I trigger the Off/On button from a script ?
I want to trigger the recording when I press an Amazon Dash Button (I know how to make the Dash Button trigger a script, but I am looking for the command to trigger)

Thanks guys

cedricve, 2018-06-19:
Yeah makes sense. We'll look into the feature request.

LBrianG, 2018-07-24:
I was able to accomplish a similar task with the API provided on the Kerberos.io website and RESTful switch component in home assistant. I have a switch on the frontend that will turn on and off recordings and an automation that turns on recording when I arm the alarm and off when I disarm. I have not worked with the dash buttons before but hopefully this can get some ideas going.