kerberos-io/machinery #155: stream freezes

marcel31415, 2018-06-18:
Hi,
the stream shows a picture from sunday (not sure from when exactly) but it still takes pictures and videos as he is supposed to do. So everything is fine except the stream.

Raspberry Pi 3B (**NOT PLUS**)
KiOS | v2.7.1 - raspberrypi3 | 
Web | 2.5.1 | 
Machinery | 2.6.1 | 

2.7.1 was flashed, so no webupdate

cedricve, 2018-06-18:
how long has it been running @marcel31415 ?

marcel31415, 2018-06-18:
System is online for 1 days, 19 hours, 24 minutes, 43 seconds

marcel31415, 2018-06-20:
System is online for 1 days, 19 hours, 37 minutes, 57 seconds 
I rebootet after I had reported this error and is working fine. I will keep an eye on that.

cedricve, 2018-06-20:
Ok thank you @marcel31415

marcel31415, 2018-06-22:
frozen, what information do you need?

Capturing works. All pictures and videos are readable and show the correct situation.

(System is online for 3 days, 21 hours, 20 minutes, 31 seconds)

cedricve, 2018-06-22:
ok, if you can send the system logs?

marcel31415, 2018-06-22:
[kerberosio-configuration.tar.gz](https://github.com/kerberos-io/machinery/files/2127459/kerberosio-configuration.tar.gz)

cedricve, 2018-06-24:
hmm, not sure what went wrong. It looks like you've restarted the instance a lot, and you also accessed the livestream over 17.000 times. Is this correct?