kerberos-io/machinery #164: system.log and daemon.log

loki-sft, 2018-07-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System and Device (please complete the following information):**
 - Host system: Linux VM, Raspberry Pi Version: [0, 1, 2, 3]
 - OS: KiOS, Raspbian, Docker, Ubuntu, CentOS, ..
 - Installation Method: KiOS installer, Etcher, Generic Installation, Docker

**Additional context**
Add any other context about the problem here.

cedricve, 2018-08-19:
hmmm, did you change any other detection settings?

loki-sft, 2018-08-20:
This is my config.xml:

`
<?xml version="1.0"?>
<kerberos>

    <instance>
        <name type="text">Garten</name>
        <logging type="bool">false</logging>
        <timezone type="timezone">Europe-Berlin</timezone>
        <capture file="capture.xml">IPCamera</capture>
        <stream file="stream.xml">Mjpg</stream>
        <condition file="condition.xml" type="multiple">Enabled</condition>
        <algorithm file="algorithm.xml">DifferentialCollins</algorithm>
        <expositor file="expositor.xml">Hull</expositor>
        <heuristic file="heuristic.xml">Sequence</heuristic>
        <io file="io.xml" type="multiple">Disk,Script</io>
        <cloud file="cloud.xml">S3</cloud>
    </instance>

</kerberos>
`

`
<?xml version="1.0"?>
<algorithms>

	<DifferentialCollins>
		<erode type="number">20</erode>
    	<threshold type="number">30</threshold>
    </DifferentialCollins>
    
	<BackgroundSubtraction>
		<shadows type="text">true</shadows>
		<history type="number">15</history>
		<nmixtures type="number">5</nmixtures>
		<ratio type="number">1</ratio>
		<erode type="number">6</erode>
		<dilate type="number">7</dilate>
    	<threshold type="number">10</threshold>
    </BackgroundSubtraction>
    
</algorithms> 
`

`<?xml version="1.0"?>
<conditions>

    <Time>
        <times type="timeselection">0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59</times>
        <delay type="number">10000</delay>
    </Time>
    <Enabled>
    	<active type="bool">true</active>
        <delay type="number">00</delay>
    </Enabled>
    
</conditions> 
`

`<?xml version="1.0"?>
<heuristics>

	<Sequence>
	    <minimumChanges type="number">20</minimumChanges>
	    <minimumDuration type="number">1</minimumDuration>
        <noMotionDelayTime type="number">1000</noMotionDelayTime>
	</Sequence>
    
	<Counter>
	    <appearance type="number">3</appearance>
	    <maxDistance type="number">140</maxDistance>
	    <minArea type="number">200</minArea>
	    <onlyTrueWhenCounted type="bool">true</onlyTrueWhenCounted>
	    <minimumChanges type="number">20</minimumChanges>
        <noMotionDelayTime type="number">100</noMotionDelayTime>
		<markers type="twolines">100,100|100,200|200,100|200,200</markers>
	</Counter>

</heuristics> `

`<?xml version="1.0"?>
<ios>

	<Disk>
        <fileFormat type="text">timestamp_microseconds_instanceName_regionCoordinates.jpg</fileFormat>
        <directory type="text">/etc/opt/kerberosio/capture/</directory>
        <markWithTimestamp type="bool">false</markWithTimestamp>
        <timestampColor type="text">white</timestampColor>
        <privacy type="bool">false</privacy>
				<throttler type="number">0</throttler>
    </Disk>
    <Video>
        <fps type="number">10</fps>
        <recordAfter type="number">5</recordAfter>
        <maxDuration type="number">30</maxDuration>
        <extension type="number">mp4</extension>
        <codec type="number">h264</codec>
        <fileFormat type="text">timestamp_microseconds_instanceName_regionCoordinates</fileFormat>
        <directory type="text">/etc/opt/kerberosio/capture/</directory>
				<hardwareDirectory type="text">/etc/opt/kerberosio/h264/</hardwareDirectory>
				<enableHardwareEncoding type="bool">true</enableHardwareEncoding>
        <markWithTimestamp type="bool">false</markWithTimestamp>
        <timestampColor type="text">white</timestampColor>
        <privacy type="bool">false</privacy>
				<throttler type="number">0</throttler>
    </Video>
    <GPIO>
        <pin type="number">17</pin>
        <periods type="number">1</periods>
        <periodTime type="number">100000</periodTime>
				<throttler type="number">0</throttler>
    </GPIO>
    <TCPSocket>
        <server type="number">127.0.0.1</server>
        <port type="number">1337</port>
        <message type="text">motion-detected</message>
				<throttler type="number">0</throttler>
    </TCPSocket>
    <Webhook>
        <url type="text">http://localhost/api/v1/webhook</url>
				<throttler type="number">0</throttler>
    </Webhook>
    <Script>
        <path type="text">/etc/opt/kerberosio/scripts/run.sh</path>
				<throttler type="number">0</throttler>
    </Script>
    <MQTT>
        <secure type="bool">false</secure>
        <verifycn type="bool">false</verifycn>
				<server type="number">127.0.0.1</server>
        <port type="number">1883</port>
				<topic type="text">kios/mqtt</topic>
        <username type="text"></username>
        <password type="text"></password>
				<throttler type="number">0</throttler>
    </MQTT>

</ios> `

neutralvibes, 2019-05-15:
Still no response to this issue? It it still happening. I counted circa 60 entries per second being logged, which negatively hits performance, when turning processing off should really do the opposite.

If your turn the machinery off, conditions should not really be tested to have to be ignored. It would be great if this could be sorted out as there are times when it is desirable to do so.

cmilanf, 2020-06-24:
It is still happening and a bit annoying as it is a waste of disk I/O and CPU. Is it not posible to change the logging level on Machinery?

cedricve, 2020-06-24:
Thanks for reporting not yet fixed, we will add a fix.