kerberos-io/machinery #165: What to run on Raspi with pi camera?

wicadmin, 2018-08-16:
So, I understand the architecture a bit and would like to run the the raspi and pi camera module in an enclosure but not run the machinery and web on it (it is a first gen pi 256MB). I prefer to run those on docker on a central server. What is recommended to run on the Pi to make it enough of an IP cam to talk to the docker components on a central docker server?

cedricve, 2018-08-19:
Interesting, you should convert the raspberry pi camera to a RTSP stream, afterwards you'll be able to enter stream url into kerberos.io

http://www.raspberry-projects.com/pi/pi-hardware/raspberry-pi-camera/streaming-video-using-vlc-player

wicadmin, 2018-08-20:
@cedricve - very good. I assume this is the best option for this board? I have several as I had bought them a while back (several years) to use a security cameras. I was following [this](https://github.com/silvanmelchior/RPi_Cam_Web_Interface) and it runs well but now would like to give kerberos.io a go instead.

With the above setup, I assume this means that data would continuously be streaming. How much Mbps do you estimate with 1080p resolution? When I thought about this, then I think it was better to run the machinery and web on the Pi but concern how it would run on it if at all?

wicadmin, 2018-09-06:
I'm receiving this error:
`06/09/2018 22:51:40.892 ERROR [trivial] OpenCV : can't open url of ip camera`

wicadmin, 2018-09-09:
OK, I have had success with [v4l2rtspserver](https://github.com/mpromonet/v4l2rtspserver). I wrote a [wiki page](https://github.com/mpromonet/v4l2rtspserver/wiki/Setup-on-Pi) there on my setup to auto start etc.

closed 2018-09-09 by wicadmin

cedricve, 2018-09-09:
cool thanks for sharing! would you mind to contribute to our documentation repo? https://github.com/kerberos-io/documentation

wicadmin, 2018-09-09:
No worries. It can be contributed.

wicadmin, 2018-09-09:
PR #12 create. 
Add documentation for RTSP server #12

Thanks.