kerberos-io/machinery #171: No stream, is the machinery running?

gieljnssns, 2018-09-12:
**Describe the bug**
I can't see my raspberry pi camera in the web view or trough the mpeg stream.
I use a raspberry pi zero W and a official raspberry pi NOIR camera

**To Reproduce**
Steps to reproduce the behavior:
1. When I stop the service
2. start kerberosio from command line
3. See error
```
pi@PiZero-Keuken:/usr/bin $ kerberosio
Bad file [/tmp/logs/myeasylog.log]
Bad file [/tmp/logs/myeasylog.log]
Bad file [/tmp/logs/myeasylog.log]
Bad file [/tmp/logs/myeasylog.log]
12/09/2018 18:26:41.410 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
12/09/2018 18:26:41.420 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
12/09/2018 18:26:41.430 INFO  [trivial] Final configuration:
- algorithm = DifferentialCollins
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = RaspiCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 500
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://admin:888888@192.168.0.13/tcp/av0_1
- captures.RaspiCamera.angle = 270
- captures.RaspiCamera.brightness = 50
- captures.RaspiCamera.contrast = 0
- captures.RaspiCamera.delay = 1000
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.RaspiCamera.framerate = 5
- captures.RaspiCamera.saturation = 0
- captures.RaspiCamera.sharpness = 0
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.fourcc = MJPG
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket =
- clouds.S3.folder =
- clouds.S3.privateKey =
- clouds.S3.publicKey =
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 10000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- configuration = /etc/opt/kerberosio/config/config.xml
- expositor = Hull
- expositors.Hull.region =
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 4
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Disk,Video,MQTT
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = true
- ios.Disk.privacy = false
- ios.Disk.throttler = 0
- ios.Disk.timestampColor = red
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.GPIO.throttler = 0
- ios.MQTT.clientId =
- ios.MQTT.password = stamnummer1
- ios.MQTT.port = 1883
- ios.MQTT.secure = false
- ios.MQTT.server = 192.168.0.54
- ios.MQTT.throttler = 8
- ios.MQTT.topic = kios/keuken
- ios.MQTT.username = hass
- ios.MQTT.verifycn = false
- ios.Pushbullet.throttler = 10
- ios.Pushbullet.token = o.mC5LPVCvPCphtSsEgWZQpFM86w9ciWQ3
- ios.Pushbullet.url = https://api.pushbullet.com
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.Script.throttler = 0
- ios.TCPSocket.message = motion-detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.TCPSocket.throttler = 0
- ios.Video.codec = h264
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.enableHardwareEncoding = true
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 30
- ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 30
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.throttler = 0
- ios.Video.timestampColor = white
- ios.Webhook.throttler = 0
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = false
- name = Keuken
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 5
- streams.Mjpg.password =
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- streams.Mjpg.username =
- timezone = Europe-Brussels
12/09/2018 18:26:41.442 INFO  [trivial] Logging is set to info
12/09/2018 18:26:41.445 INFO  [trivial] Capture: Start capture device: RaspiCamera
12/09/2018 18:26:41.447 INFO  [trivial] Capture: Trying to open Raspberry Pi camera module.
12/09/2018 18:26:41.448 INFO  [trivial] Capture: (Warning) If you see a OMX_GetHandle error, this means that you don't have a working RPi camera module attached.
12/09/2018 18:26:41.449 INFO  [trivial] Capture: (Warning) You can change the capture device with the configuration files.
CopyPort( OMX.broadcom.camera->70, OMX.broadcom.video_encode->200
Port 200: in 1/1 460800 16 disabled,not pop.,not cont. 640x480 640x480 @5fps 20
===> types : 8[200] | 8[71]
Port 71: out 1/1 460800 16 disabled,not pop.,not cont. 640x480 640x480 @5fps 20
Port 200: in 1/1 15360 16 disabled,not pop.,not cont. 160x64 160x64 @30fps 20
[OMX.broadcom.camera]Wating state to be 2
[OMX.broadcom.video_encode]Wating state to be 2
[OMX.broadcom.video_encode]Wating state to be 2
Wating port 70 to be 1
Wating port 200 to be 1
Port 70: out 1/1 460800 16 enabled,populated,not cont. 640x480 640x480 @5fps 20
Wating port 71 to be 1
Port 71: out 0/0 460800 16 enabled,populated,not cont. 640x480 640x480 @5fps 20
[OMX.broadcom.camera]Wating state to be 3
Wating port 201 to be 1
Port 200: in 1/1 460800 16 enabled,populated,not cont. 640x480 640x480 @5fps 20
Port 201: out 1/1 65536 16 enabled,populated,not cont. 640x480 0x0 @30fps 0
[OMX.broadcom.video_encode]Wating state to be 3
Wating port 201 to be 1
Wating port 200 to be 1
Port 200: in 0/0 460800 16 enabled,populated,not cont. 640x480 640x480 @5fps 20
Port 201: out 1/1 65536 16 enabled,populated,not cont. 640x480 0x0 @30fps 0
[OMX.broadcom.video_encode]Wating state to be 3
12/09/2018 18:26:42.693 INFO  [trivial] Capture: Succesfully opened Raspberry Pi camera module.
12/09/2018 18:26:47.695 INFO  [trivial] Capture: Start streaming thread
12/09/2018 18:26:47.697 INFO  [trivial] Stream: Configured stream on port 8889 with quality: 75
12/09/2018 18:26:47.698 INFO  [trivial] Starting cloud service: S3
12/09/2018 18:26:47.818 INFO  [trivial] Starting conditions: Enabled
12/09/2018 18:26:47.831 INFO  [trivial] Starting algorithm: DifferentialCollins
12/09/2018 18:26:47.835 INFO  [trivial] Starting expositor: Hull
OpenCV Error: Assertion failed (total >= 0 && (depth == CV_32S || depth == CV_32F)) in pointPolygonTest, file /home/cedricverst/kios/output/raspberrypi/build/kerberosio-machinery-develop/opencv/modules/imgproc/src/geometry.cpp, line 103
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/cedricverst/kios/output/raspberrypi/build/kerberosio-machinery-develop/opencv/modules/imgproc/src/geometry.cpp:103: error: (-215) total >= 0 && (depth == CV_32S || depth == CV_32F) in function pointPolygonTest

Aborted
```

**Expected behavior**
I would like to see the stream

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Operating System and Device (please complete the following information):**
 - Host system: Raspberry Pi Zero W
 - OS: Raspbian
 - Installation Method: Generic Installation

**Additional context**
Add any other context about the problem here.

Shboodle, 2019-06-27:
Hi,

I´ve the same issue.
After 30m the module stop working. Error below:
"OMX_GetHandle error, this means that you don't have a working RPi camera module attached"
Can you please help us?

Thank you
Tiago

doctorpepper608, 2019-07-03:
Hi, i have same problem. And after rebuild ffmpeg to lowest version and rebuild machinery I have received "Segmentation fault "