kerberos-io/machinery #175: v2.6.2/armbian-machinery-kerberosio-armhf-2.6.2.deb not found

supersh86, 2018-09-21:
missing file installer for armbian
https://github.com/kerberos-io/machinery/releases/download/v2.6.2/armbian-machinery-kerberosio-armhf-2.6.2.deb

cedricve, 2018-11-06:
Can someone compiled the armbian deb file for us?

jsunwalters, 2019-01-02:
I compiled from source and created this .deb file.  It is my first so please let me know if it works for you or not.
[Armbian-2.6.2-machinery.zip](https://github.com/kerberos-io/machinery/files/2721762/Armbian-2.6.2-machinery.zip)

cedricve, 2019-01-02:
thanks @jsunwalters, can you verify @supersh86

tloki, 2019-04-28:
Is it safe to use this binary some random person posted on internet? Especially when he is not a contributor not has really any history on github? Not accusing anyone - but from a security perspective my guts tell me to not use it before official release is available...

jsunwalters, 2019-04-29:
I totally understand your concern.  What can I do to verify my
authenticity?  I am a long time computer guy but did most of my programming
back in the C and BSD 4.3 days lol.  We barely had internet let alone
github. I started to get into home automation which is what led me back
into programming and to kerberos-io.  I needed this version so I compiled
it.

My work address is jason.walters@hpe.com if you want to email me there for
confirmation I am legit?  Not sure what else to offer.

Jason

tloki, 2019-04-29:
Your version indeed does work on latest armbian as of May 2019. (ARMBIAN 5.75 stable Ubuntu 18.04.2 LTS 4.19.25-sunxi) for Bananapi M1 - but I'll put the deployment on hold since no official binary is released.

It's not a problem that I don't believe you, but the fact that there is some binary posted on internet which is not official kerberos.

Probably build script will do the job (even better if in docker image or some other redistributable form).

About armbian and kerberos - I had plenty of things to be done no matter the documentation available here:

https://doc.kerberos.io/2.0/installation/Armbian

for 5.75 instead of

`sudo apt-get update && sudo apt-get install libav-tools`

one should install the following:

`sudo apt update && sudo apt install -y libcurl3 libcurl-openssl1.0-dev ffmpeg`

for web - one should install gpg raspbian repo key since armbian refuses to install from it unsigned:

`echo "deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi" | sudo tee --append /etc/apt/sources.list
sudo apt-get update && sudo apt-get upgrade
gpg --keyserver keys.gnupg.net --recv-key 9165938D90FDDD2E
gpg -a --export 9165938D90FDDD2E | sudo apt-key add -`

also - there should be a documentation on how to set up HTTPS using letsencrypt (take a look at nextcloud tutorials for example). Here's what one would need to do:

(first port forward your 80 and 443 ports and make sure that you can access your kerberos http server. no need to login yet since the password will not be encrypted)

sudo apt install python3-pip python3-setuptools python3-dev libffi-dev build-essential gcc make g++
sudo -H pip3 install wheel
sudo -H pip3 install certbot
sudo -H pip3 install certbot-nginx

sudo certbot --nginx --agree-tos --redirect --hsts --staple-ocsp --email your-email-address -d forexamplekerberos.mydomain.letssaycom

you may now test it:
https://www.ssllabs.com/ssltest/analyze.html?d=forexamplekerberos.mydomain.letssaycom

edit
/etc/nginx/sites-enabled/default as root and delete server entry that was not managed by certbot (for me it was the first one,
and two more were created by certbot)

add some headers to the bigger server entry (443 one, not 80 one):

add_header X-Content-Type-Options nosniff;
add_header X-XSS-Protection "1; mode=block";
add_header X-Robots-Tag none;
add_header X-Download-Options noopen;
add_header X-Permitted-Cross-Domain-Policies none;

and to renew certificates:
sudo certbot renew to renew

this should probably be separate issue / pull request for 2.0 armbian (>5.25 I assume...) installation, as well as HTTPS using letsencrypt setup (it's 2019...).

cedricve, 2019-05-15:
@jsunwalters @tloki, thank you for your help on this! Highly appreciated.

@tloki can you confirm with the changes made (mentioned above), you got it running? If so are you interested in sending a PR to the documentation repo. As I believe you are best in place for helping us with this.

Let me know if you can help. thanks all!