kerberos-io/machinery #176: Working camera resolutions are limited to 1280x960

mtchin, 2018-11-02:
**Describe the bug**
Working camera resolutions are limited to 1280x960.
Get RaspberryPi camera module v2 (5MP 2592x1944) and it is not supported, also on ZeroW it is extremely slow.

**To Reproduce**
**Expected behavior**
Clear pictures and video

**Screenshots**

**Operating System and Device (please complete the following information):**
 - Host system: Raspberry Pi Version: 0 W
 - OS: KiOS 
 - Installation Method: Etcher

**Additional context**

mtchin, 2019-01-17:
Any news on this ? Anyone with the same issue ?

KarlBaumann, 2020-02-13:
I guess I am having the same issue. Everything larger than that looks like this 
[image]
the weird part is that the live stream looks good.