kerberos-io/machinery #178: Document the options for file format fully

estechnical, 2018-12-01:
**Is your feature request related to a problem? Please describe.**
I am configuring the machinery docker image without using the web container, I'd really like to be able to refer to a document that details all the options that I can configure. Particularly of interest would be the <Video><fileFormat> ... options. I don't find a good explanation of what I can set here and I'd like to change the file format to a human readable date format.

**Describe the solution you'd like**
Describe the options thoroughly in some reference documents.

**Describe alternatives you've considered**
I am currently looking at the web container and examining how these options are configured. This isn't quick and easy to refer to.

estechnical, 2018-12-01:
I guess I could rephrase this as "how do I configure machinery to output files like so?"

<capture path>/[YYYY-MM-DD]/[HH-mm-ss].mp4

cpatenta, 2019-06-30:
I am also interested in this. I tried to change video files name to include date and time with no success.
Thank you!

Shommes1981, 2022-03-18:
Is there any update on this? Am interested also