kerberos-io/machinery #179: Unable to install Kerberosio Generic

robinrego, 2018-12-03:
Unable to install Kerberosio Generic
OS: Ubuntu Server 18.04
Device: intel core i5 ; 4 gb ram
---

I have followed instrusction on https://doc.kerberos.io/2.0/installation/Generic
and get stuck at:

Start the machinery on start-up.
sudo systemctl enable kerberosio

error: Failed to enable unit: Unit file Kerberosio.service does not exist.

I tried the installation several times and it seems the machinary is not getting installed.
Please let me know what I am missing.

---
robin@kerberosio:~$ locate kerberosio
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/DependInfo.cmake
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/build.make
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/cmake_clean.cmake
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/depend.make
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/flags.make
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/link.txt
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/progress.make
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos/capture
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos/cloud
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos/machinery
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos/machinery/algorithm
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos/machinery/condition
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos/machinery/expositor
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos/machinery/heuristic
/home/robin/machinery/build/src/CMakeFiles/kerberosio.dir/kerberos/machinery/io
/home/robin/machinery/service/kerberosio.service
/home/robin/machinery/src/kerberosio.icns
robin@kerberosio:~$
---

cedricve, 2018-12-03:
@robinrego did you do make install ?

robinrego, 2018-12-03:
I did it now and it works !
I thought it was done in "cmake .. && make && make check && sudo make install"
Thank you

closed 2018-12-03 by robinrego