kerberos-io/machinery #185: Loop when enable MQTT action

jdavisonc, 2019-01-07:
Machinery start loop once I configure MQTT actions, If I disabled it all goes to normal.

```
7/01/2019 23:47:19.538 INFO  [trivial] Logging is set to verbose
07/01/2019 23:47:19.538 INFO  [trivial] Capture: Start capture device: IPCamera
07/01/2019 23:47:19.539 INFO  [trivial] Capture: Trying to open IP camera.
07/01/2019 23:47:19.539 INFO  [trivial] Capture: (Warning) You can change the capture device with the configuration files.
07/01/2019 23:47:21.455 INFO  [trivial] Capture: Succesfully opened IP camera.
07/01/2019 23:47:21.456 INFO  [business] Capture: Start capture grab thread
07/01/2019 23:47:21.456 INFO  [business] Capture: Start capture health thread
07/01/2019 23:47:26.456 INFO  [trivial] Capture: Start streaming thread
07/01/2019 23:47:26.459 INFO  [business] Capture: checking health status of camera.
07/01/2019 23:47:26.459 INFO  [trivial] Stream: Configured stream on port 8889 with quality: 75
07/01/2019 23:47:26.459 INFO  [trivial] Starting cloud service: S3
07/01/2019 23:47:26.472 INFO  [trivial] Starting conditions: Enabled,Enabled
07/01/2019 23:47:26.472 INFO  [trivial] Starting algorithm: BackgroundSubtraction
07/01/2019 23:47:26.472 INFO  [trivial] Starting expositor: Hull
07/01/2019 23:47:26.513 INFO  [trivial] Starting heuristic: Sequence
07/01/2019 23:47:26.513 INFO  [trivial] Starting io devices: Video,MQTT
07/01/2019 23:47:27.613 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
07/01/2019 23:47:27.615 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
07/01/2019 23:47:27.617 INFO  [trivial] Final configuration:
- algorithm = BackgroundSubtraction
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 6
- algorithms.BackgroundSubtraction.history = 200
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 0.9
- algorithms.BackgroundSubtraction.shadows = true
- algorithms.BackgroundSubtraction.threshold = 200
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 250
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://192.168.86.135:554/ch0_0.h264
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.brightness = 50
- captures.RaspiCamera.contrast = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.RaspiCamera.framerate = 20
- captures.RaspiCamera.saturation = 0
- captures.RaspiCamera.sharpness = 0
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket = 
- clouds.S3.folder = 
- clouds.S3.privateKey = 
- clouds.S3.publicKey = 
- condition = Enabled,Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 2000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:30,6:30-0:30,6:30-0:30,6:30-0:30,6:30-0:30,6:30-0:30,6:30-0:30,6:30
- configuration = /etc/opt/kerberosio/config/config.xml
- expositor = Hull
- expositors.Hull.region = 448,463|548,184|485,143|140,172|75,96|7,139|15,462
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 5
- heuristics.Counter.markers = 931,390|473,173|814,634|56,309
- heuristics.Counter.maxDistance = 300
- heuristics.Counter.minArea = 100
- heuristics.Counter.minimumChanges = 15
- heuristics.Counter.noMotionDelayTime = 500
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 50
- heuristics.Sequence.minimumDuration = 10
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Video,MQTT
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.throttler = 0
- ios.Disk.timestampColor = none
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.GPIO.throttler = 0
- ios.MQTT.password = admin
- ios.MQTT.port = 1883
- ios.MQTT.secure = true
- ios.MQTT.server = mosquitto
- ios.MQTT.throttler = 10
- ios.MQTT.topic = kerberos/frontdoor
- ios.MQTT.username = admin
- ios.MQTT.verifycn = true
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.Script.throttler = 0
- ios.TCPSocket.message = motion-detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.TCPSocket.throttler = 0
- ios.Video.codec = h264
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.enableHardwareEncoding = true
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 30
- ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
- ios.Video.markWithTimestamp = true
- ios.Video.maxDuration = 30
- ios.Video.privacy = false
- ios.Video.recordAfter = 2
- ios.Video.throttler = 0
- ios.Video.timestampColor = w^C

```

cedricve, 2019-05-09:
it's been a while but did you found the root cause?

Stubbs, 2019-12-21:
I'm having the same issue.

```
- ios.MQTT.password = ....
- ios.MQTT.port = 1883
- ios.MQTT.secure = false
- ios.MQTT.server = mqtt
- ios.MQTT.throttler = 10
- ios.MQTT.topic = kerberos/lounge
- ios.MQTT.username = ....
- ios.MQTT.verifycn = false
```

There's nothing in the MQTT logs to suggest it's actually making a connection, I have other mqtt clients connecting with these credentials and I can ping the hostname I'm using for the server. I've even tried the IP of the host.

In Kerberos' logs the error loop I'm getting is:

```
lounge-cctv                 | 2019-12-21 21:02:25,436 INFO exited: machinery (terminated by SIGABRT (core dumped); not expected)
lounge-cctv                 | 2019-12-21 21:02:26,438 INFO spawned: 'machinery' with pid 2545
lounge-cctv                 | 2019-12-21 21:02:27,519 INFO success: machinery entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
```

I'm running Kerberos under Docker as well.

I notice on @jdavisonc original post that it's set to use SSL to connect to mqtt, but on the non-ssl port, and cn verification is left on too, most default certs would fail that check.