kerberos-io/machinery #188: trying to install kerberosio results in this error when i "make check"

terny1, 2019-02-08:
trying to install kerberosio results in this error when i "make check"
using this https://doc.kerberos.io/2.0/installation/Generic

i type: cd machinery && mkdir build && cd build
then: cmake .. && make
then: make check
then comes the error:
`[ 59%] Built target ENCODE_LIBRARY
[ 59%] Built target googletest
[ 59%] Built target restclient
[ 59%] Built target mosquitto
[ 59%] Built target opencv
[ 59%] Built target TINYXML_LIBRARY
[ 59%] Built target EXECUTOR_LIBRARY
[ 59%] Built target FILEWATCHER_LIBRARY
[ 59%] Built target KERBEROS_CORE
[ 59%] Linking CXX executable ../../test/bin/kerberos-test
/home/terny/anaconda3/lib/libpng.so: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
test/CMakeFiles/kerberos-test.dir/build.make:772: recipe for target '../test/bin/kerberos-test' failed
make[3]: *** [../test/bin/kerberos-test] Error 1
CMakeFiles/Makefile2:1580: recipe for target 'test/CMakeFiles/kerberos-test.dir/all' failed
make[2]: *** [test/CMakeFiles/kerberos-test.dir/all] Error 2
CMakeFiles/Makefile2:1547: recipe for target 'test/CMakeFiles/check.dir/rule' failed
make[1]: *** [test/CMakeFiles/check.dir/rule] Error 2
Makefile:702: recipe for target 'check' failed
make: *** [check] Error 2
`

jradxl, 2019-03-02:
I had this a few moments (well, hours) ago..... I also noticrd that ffmpeg did not run - missing a library...

In trying to faultfind, I looked at ....
https://trac.ffmpeg.org/wiki/CompilationGuide/Ubuntu
My intention was to see if the latest version would build...
I followed the those instructions to apt-get everything.
I also rebooted...
I also did a cmake .. make and make install again
Now make check works...
Not sure what fixed the issue - perhaps the reboot, perhaps the apt-gets, perhaps the rebuild.
Perhaps you could try a different combination.

cedricve, 2019-03-02:
hey @jradxl, indeed ffmpeg is a required library. If you would run kerberosio from the commandline you will see the dependency issue. Good that you resolved the issue yourself. You can close the issue ;)

closed 2019-05-09 by cedricve