kerberos-io/machinery #190: Random crashes, device left in blocking state (?), needs reboot

nopmop, 2019-02-25:
**Description**
Everything works with default settings, but when I set the option to call a bash script `kerberosio` keeps crashing (being respawned by `systemd`). Reverting back the config doesn't work (settings are not stored anymore), and running the process manually shows only the following:

```
[...]
25/02/2019 22:36:21.856 INFO  [trivial] Capture: Succesfully opened Raspberry Pi camera module.
25/02/2019 22:36:26.857 INFO  [trivial] Capture: Start streaming thread
25/02/2019 22:36:26.858 INFO  [trivial] Stream: Configured stream on port 8889 with quality: 75
25/02/2019 22:36:26.858 INFO  [trivial] Starting cloud service: S3
25/02/2019 22:36:26.886 INFO  [trivial] Starting conditions: Enabled
25/02/2019 22:36:26.887 INFO  [trivial] Starting algorithm: DifferentialCollins
25/02/2019 22:36:26.888 INFO  [trivial] Starting expositor: Hull
25/02/2019 22:36:27.761 INFO  [trivial] Starting heuristic: Sequence
25/02/2019 22:36:27.761 INFO  [trivial] Starting io devices: Video
25/02/2019 22:36:31.859 INFO  [trivial] Capture: devices is blocking, and not grabbing any more frames.
terminate called after throwing an instance of 'kerberos::KerberosCouldNotGrabFromCamera'
  what():  Kerberos : Capture : Could not grab from capture: devices is blocking, and not grabbin any more frames.
Aborted
```
After reboot & config revert things start to operate normally again. It looks like after the first crash (maybe happening for trivial reasons) a deadlock is happening because the camera is left in an unclean state.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `Configuration` -> `Motion` --- (click next screen * 2) ---> enable `SCRIPT` output

**Expected behavior**
Also strange that the camera seems to be left in an unclean state and that only a reboot fixes it.

**Operating System and Device:**
 - Host system: Raspberry Pi Version: 3
 - OS: Raspbian
 - Installation Method: Generic Installation

**Additional context**
- No other spurious processes are using any network/files/devices required for kerberosio to operate.
- The bash script set to be called has correct execution & access permissions.

A strace shows the following:
```
[...]

[pid  2409] open("/etc/opt/kerberosio/symbols/", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_DIRECTORY|O_CLOEXEC) = 15
[pid  2409] fstat64(15, {st_mode=S_IFDIR|0777, st_size=4096, ...}) = 0
[pid  2409] getdents64(15, /* 2 entries */, 32768) = 48
[pid  2409] getdents64(15, /* 0 entries */, 32768) = 0
[pid  2409] close(15)                   = 0
[pid  2409] nanosleep({tv_sec=1, tv_nsec=500000000},  <unfinished ...>
[pid  2392] write(14, "\3676\254t\326\271\262\vtcBI\303\250\245\226\322]1\362J\356u\332\230\31\r\376'\332\254\331"..., 4096) = 4096
[pid  2392] write(14, "NZ+\235\321]\311-lY\201DA#/>i\253\321\332\317\252\333}\2322\335Aun\v}"..., 4096) = 4096
[pid  2392] write(14, "\320v\372\326\25[E\246\306;\313n\301\0262\3ch\307\31\364>\265>\234\261\220\f\222\365S\204"..., 1824) = 1824
[pid  2392] close(14)                   = 0
[pid  2392] inotify_init()              = 14
[pid  2392] inotify_add_watch(14, "/etc/opt/kerberosio/config", IN_CLOSE_WRITE|IN_MOVED_FROM|IN_MOVED_TO|IN_CREATE|IN_DELETE) = 1
[pid  2392] _newselect(15, [14], NULL, NULL, {tv_sec=0, tv_usec=0}) = 0 (Timeout)
[pid  2392] mmap2(NULL, 5763072, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x656b9000
[pid  2404] <... nanosleep resumed> NULL) = 0
[pid  2404] nanosleep({tv_sec=0, tv_nsec=33300000},  <unfinished ...>
[pid  2406] <... nanosleep resumed> NULL) = 0
[pid  2406] _newselect(10, [9], NULL, NULL, {tv_sec=0, tv_usec=10}) = 0 (Timeout)
[pid  2406] nanosleep({tv_sec=0, tv_nsec=33333000},  <unfinished ...>
[pid  2392] mmap2(NULL, 1921024, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x6adbe000
[pid  2392] futex(0x251024c, FUTEX_WAKE_OP_PRIVATE, 1, 1, 0x2510248, FUTEX_OP_SET<<28|0<<12|FUTEX_OP_CMP_GT<<24|0x1) = 1
[pid  2432] <... futex resumed> )       = 0
[pid  2392] futex(0x25102ac, FUTEX_WAKE_OP_PRIVATE, 1, 1, 0x25102a8, FUTEX_OP_SET<<28|0<<12|FUTEX_OP_CMP_GT<<24|0x1) = 1
[pid  2433] <... futex resumed> )       = 0
[pid  2392] futex(0x251030c, FUTEX_WAKE_OP_PRIVATE, 1, 1, 0x2510308, FUTEX_OP_SET<<28|0<<12|FUTEX_OP_CMP_GT<<24|0x1) = 1
[pid  2392] futex(0x251036c, FUTEX_WAKE_OP_PRIVATE, 1, 1, 0x2510368, FUTEX_OP_SET<<28|0<<12|FUTEX_OP_CMP_GT<<24|0x1) = 1
[pid  2392] futex(0x25103cc, FUTEX_WAKE_OP_PRIVATE, 1, 1, 0x25103c8, FUTEX_OP_SET<<28|0<<12|FUTEX_OP_CMP_GT<<24|0x1) = 1
[pid  2392] futex(0x251042c, FUTEX_WAKE_OP_PRIVATE, 1, 1, 0x2510428, FUTEX_OP_SET<<28|0<<12|FUTEX_OP_CMP_GT<<24|0x1) = 1
[pid  2437] <... futex resumed> )       = 0
[... 222 lines cut ...]
[pid  2437] +++ killed by SIGABRT +++
[pid  2439] +++ killed by SIGABRT +++
[pid  2435] +++ killed by SIGABRT +++
[pid  2433] +++ killed by SIGABRT +++
[pid  2432] +++ killed by SIGABRT +++
[pid  2436] +++ killed by SIGABRT +++
[pid  2431] +++ killed by SIGABRT +++
[pid  2430] +++ killed by SIGABRT +++
[pid  2414] +++ killed by SIGABRT +++
[pid  2413] +++ killed by SIGABRT +++
[pid  2409] +++ killed by SIGABRT +++
[pid  2408] +++ killed by SIGABRT +++
[pid  2406] +++ killed by SIGABRT +++
[pid  2404] +++ killed by SIGABRT +++
[pid  2403] +++ killed by SIGABRT +++
[pid  2402] +++ killed by SIGABRT +++
[pid  2401] +++ killed by SIGABRT +++
[pid  2400] +++ killed by SIGABRT +++
[pid  2399] +++ killed by SIGABRT +++
[pid  2398] +++ killed by SIGABRT +++
[pid  2397] +++ killed by SIGABRT +++
[pid  2396] +++ killed by SIGABRT +++
[pid  2405] +++ killed by SIGABRT +++
+++ killed by SIGABRT +++
```

ChieftainY2k, 2020-04-21:
It seems I have exactly same problem with kerberos machinery in docker container.

After the configuration update I get:
`kerberos_1            | [OMX.broadcom.video_encode]Wating state to be 3
kerberos_1            | 21/04/2020 10:12:59.509 INFO  [trivial] Capture: Succesfully opened Raspberry Pi camera module.
kerberos_1            | 21/04/2020 10:13:04.510 INFO  [trivial] Capture: Start streaming thread
kerberos_1            | 21/04/2020 10:13:04.511 INFO  [trivial] Stream: Configured stream on port 8889 with quality: 75
kerberos_1            | 21/04/2020 10:13:04.511 INFO  [trivial] Starting cloud service: S3
kerberos_1            | 21/04/2020 10:13:04.528 INFO  [trivial] Starting conditions: Enabled
kerberos_1            | 21/04/2020 10:13:04.529 INFO  [trivial] Starting algorithm: DifferentialCollins
kerberos_1            | 21/04/2020 10:13:04.529 INFO  [trivial] Starting expositor: Hull
kerberos_1            | 21/04/2020 10:13:04.923 INFO  [trivial] Starting heuristic: Sequence
kerberos_1            | 21/04/2020 10:13:04.924 INFO  [trivial] Starting io devices: Video,Disk,MQTT
kerberos_1            | 21/04/2020 10:13:05.833 INFO  [trivial] HullExpositor: activity detected from (9,8) to (208,45)
kerberos_1            | 21/04/2020 10:13:09.512 INFO  [trivial] Capture: devices is blocking, and not grabbing any more frames.
kerberos_1            | /container-runner.sh: line 96: 32701 Aborted                 (core dumped) kerberosio
`

cedricve, 2020-04-21:
@ChieftainY2k  You didnt had this with the 2.6.1?

ChieftainY2k, 2020-04-21:
@cedricve 
Indeed I did have it but I thought it was some system misconfiguration on my part.  
After thorough testing I did yesterday it's present on RPI2/3/4 (I was able to test it with RPI4 which I got yesterday , yay! :) ) )

> @ChieftainY2k You didnt had this with the 2.6.1?

cedricve, 2020-04-21:
> @cedricve
> Indeed I did have it but I thought it was some system misconfiguration on my part.
> After thorough testing I did yesterday it's present on RPI2/3/4 (I was able to test it with RPI4 which I got yesterday , significant upgrade, yay! :-) )
> 
> However it does not happen every time, strange....
> 
> > @ChieftainY2k You didnt had this with the 2.6.1?

Cool ;) Does it occur when you dont use the script function?

ChieftainY2k, 2020-04-21:
> Cool ;) Does it occur when you dont use the script function?

Unfortunately yes, it does :( here's the config I use:

[kerberos.zip](https://github.com/kerberos-io/machinery/files/4508777/kerberos.zip)

ChieftainY2k, 2020-04-21:
@cedricve I can open up ssh via the ngrok tunnel for you to look inside the kerberos container to fiddle around if you want :)

cedricve, 2020-04-21:
thanks, just wondering if you are using latest version of the Raspbian OS?

ChieftainY2k, 2020-04-21:
I'm using the newest Raspbian Buster with the latest apt-get updates.

> thanks, just wondering if you are using latest version of the Raspbian OS?

ChieftainY2k, 2020-04-21:
May be unrelated but sometimes this appears in the /var/log/kern.log 

```
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571559] INFO: task kerberosio:16407 blocked for more than 120 seconds.
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571570] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571575] kerberosio      D    0 16407   2501 0x00000004
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571606] [<c09c95cc>] (schedule) from [<c09cd5b4>] (schedule_timeout+0x200/0x428)
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571622] [<c09cbb48>] (__down) from [<c0278cd0>] (down+0x54/0x68)
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571643] [<c086d590>] (vchiq_release) from [<c03d66fc>] (__fput+0xa4/0x1e8)
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571661] [<c03d68b0>] (____fput) from [<c02426dc>] (task_work_run+0xbc/0xe0)
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571679] [<c0226bd0>] (do_exit) from [<c0227434>] (do_group_exit+0x4c/0xe4)
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571695] [<c0233780>] (get_signal) from [<c020ca1c>] (do_signal+0x364/0x474)
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571709] [<c020cd38>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571718] ffa0:                                     aa434b40 00000000 00000002 00000000
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571724] ffc0: 00000018 00000000 00000000 000000a2 03b7a338 b0680860 aa434ca8 b06808d8
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571729] ffe0: 00000000 aa434b38 00000000 b633b030 80000010 aa434b40
Apr 21 15:15:14 raspberry4 kernel: [ 7739.571736] INFO: task kworker/2:1:14577 blocked for more than 120 seconds.
```