kerberos-io/machinery #192: Exposure tuning (nightpreview)

nopmop, 2019-03-01:
**Problem**
my problem is **dark images** - I'm aware of the FPS<->light relationship (as detailed at https://github.com/kerberos-io/machinery/issues/12) and of possible workarounds (like using an RTSP server as detailed here https://github.com/kerberos-io/machinery/issues/165, which hammers the CPU considerably though).

**Solution i'd like**
Being able to tune the `mmal exposure` parameter (set to `nightpreview` for example) from kerberosio would be nice. 

**Question**
Is this going to be contemplated for a future release? Where to look in the code to submit a PR?

nopmop, 2019-03-16:
Bump. Any help?

cedricve, 2019-03-18:
hey @nopmop, this is a hard one. You will need to consider looking into this library.github.com/dridri/OpenMaxIL-cpp

gbus, 2019-07-28:
Hi @cedricve 
I have just added support for setExposureControl in https://github.com/dridri/OpenMaxIL-cpp.
I have tested it from kerberos.io with this simple change, but I guess that it takes you 5 minutes to make it configurable from the raspicam settings in capture.xml:

```
diff --git a/src/kerberos/capture/RaspiCamera.cpp b/src/kerberos/capture/RaspiCamera.cpp
index 07c1a97..f17ef9c 100755
--- a/src/kerberos/capture/RaspiCamera.cpp
+++ b/src/kerberos/capture/RaspiCamera.cpp
@@ -218,7 +218,7 @@ namespace kerberos
 				state.camera->setSaturation(m_saturation);
 				state.camera->setContrast(m_contrast);
 				state.camera->setSharpness(m_sharpness);
-				//state.camera->setExposureControl(Camera::ExposureControlOff);
+				state.camera->setExposureControl(Camera::ExposureControlNight);
 
 				if(m_angle > 0)
 				{
```

Thanks

cedricve, 2019-07-28:
super cool @gbus, do you have any results already?

gbus, 2019-07-28:
Yes, with ExposureControlNight the image from a room with little light looked very bright, while with ExposureControlSnow was totally dark.
:-)

cedricve, 2019-07-28:
Cool ! and what about the recording speed, just curious. How much FPS do you get.
Will definitely check this out myself. Great job!!

gbus, 2019-07-28:
I tried 1,3,5,10,20,30 fps, where at 10fps the exposure was similar to what I could see with my bare eyes. I would consider 1, 3 and 5fps like 3 possible level of exposure improvement (and image quality).
Here are the sample pictures:

[image]
[image]
[image]
[image]
[image]
[image]