kerberos-io/machinery #195: The machinery receives a SIGABRT and this temporarily kills the camera stream.

chirag-b, 2019-05-03:
**Bug Description**
The machinery receives a SIGABRT signal and terminates. This temporarily stops the camera stream. The camera is a Panasonic WV-SFN130 POE camera. We have a kerberos container to read the RTSP stream from this camera and serve it via HTTP. We used OpenCV (3.4.3.18) for Python 3.7 to read frames from the kerberos container. The stream stops with no frame being read causing our program that is reading the HTTP stream from the container to terminate. However after a short while the machinery comes back up and everything is back to normal for a short while. This cycle repeats. Check screenshots for more info.

**PLEASE NOTE**
When I do not use Kerberos and directly access the RTSP stream, everything works fine. The stream does not die. With Kerberos this is not the case.

**To Reproduce**
Steps to reproduce the behavior:
1. You will need a Panasonic WV-SFN130 POE Camera. 
2. Setup a kerberos container and use the RTSP stream from the camera.
3. A simple python script to read the stream of the created kerberos container.

**Expected behavior**
The expectation is that the stream does not die causing our program reading the stream to terminate. Kerberos machinery should not do whatever it is doing in order to receive a SIGABRT.
OR Is there a way we can avoid having this signal being sent to the machinery? 

**Screenshots**
[image]

High CPU usage. Screenshot Below
[image]

**Operating System and Device:**
OS: Ubuntu 18.04.1 LTS 64Bit
Memory: 32GB
Processor: Intel Core i7-8700K
Camera: Panasonic WV-SFN130 POE.
Kerberos Version: commit 4b6588373f929f6a3f1cd3bb39d36558d8ce4361

**Additional context**
The CPU usage of the kerberos containers shoots up when we begin reading the stream off Kerberos container using OpenCV.

cedricve, 2019-05-09:
Additional context
The CPU usage of the kerberos containers shoots up when we begin reading the stream off Kerberos container using OpenCV.

- Yes that is because it starts to encoding the images (Kerberos.io reads the stream and converts it to a MJPEG stream to visualise it on the front-end. However why aren't you fetching the native RTSP stream from the camera?

chirag-b, 2019-05-09:
We use Kerberos to serve the camera stream from a website. Multiple clients connect to this HTTP stream. We don't want to host the RTSP stream directly from the website without Kerberos. Also I'm not very sure if there can be two or more different connections to an RTSP stream. Correct me if I'm wrong. Just another thing is that the clients are always connected to the Kerberos containers.

In any case, at some point during the streaming process (minutes later), the Kerberos container receives a SIGABRT (many times) and this causes us to lose the stream briefly. It seems like the machinery is doing something wrong such that it is receiving these signals.

closed 2021-03-12 by chirag-b