kerberos-io/machinery #198: Machinery (v3.0) move to Golang

cedricve, 2019-07-14:
We are currently working on a new major version of Kerberos.io. This will contain a full rewrite of the machinery using the Golang programming language. Features that will be included:

- Pre and post recording.
- Continuous recording.
- Multiple ip camera configuration OOTB.
- Remote reconfiguring using Kerberos.cloud
- MQTT as base protocol to send and retrieve information.
- Livestreaming using WEBRTC, to retrieve high-res and low-latency streaming.
- Change to more open license.

The plan is to focus on IP camera's first, and afterwards add support for other types.

More information to come.

lahdekorpi, 2019-08-19:
Is the development happening behind closed doors for now or has this started yet?
I may be able to contribute on the API, React or v4l2 on Go.

cedricve, 2019-08-19:
@lahdekorpi no it will not be closed, but we are working on a first version of the solution. Once released and published, we will opensource it.

Cédric

jurenovic, 2019-11-24:
Hello @cedricve, would like to know when will this be available. Thanks

cedricve, 2019-12-03:
The goal is 1th of January. We are about to revive the project and update a couple or more repo's in the meantime. More news coming soon.

Stubbs, 2019-12-21:
That explains the "disable phpunit for now" commit :-D

cedricve, 2019-12-23:
> That explains the "disable phpunit for now" commit :-D

This was just to make the build working on Travis-CI. However we will change a couple of things for the introduction in Q1.

cedricve, 2019-12-23:
> The goal is 1th of January. We are about to revive the project and update a couple or more repo's in the meantime. More news coming soon.

Short update.

We will announce some changes and the roadmap on January 1st. New solutions will be released in Q1.

cedricve, 2020-01-01:
We made some changes (documentation website) and added support for Raspberry Pi 4. Also the license has been updated to BY-NC-SA 4.0 in version 2.8.0 We will further work, and plan to release 3.0 in Q1/2020.

Keep you posted.

dklemm, 2020-08-23:
Has this been abandoned? I'd like to get involved.

cedricve, 2020-08-23:
@dklemm nope working very hard on this to be honest (still behind closed doors). No ETA for the moment, however we will further communicate on this Github issue.

cedricve, 2020-11-17:
A short update:

Pre and post recording.
Continuous recording.
Multiple camera configuration OOTB.
Remote reconfiguring using Kerberos.cloud
MQTT as base protocol to send and retrieve information.
Live streaming using WEBRTC, to retrieve high-res and low-latency streaming.
Change to more open license.
JSON confiiguration files.
ONVIF auto detection.
Adapters to send basic notifications (similar to prev releases in 2.x).
Configurable auto-removal strategy.

Currently all these features are integrated Kerberos Enterprise 1.2.0. We managed to get Kerberos Enterprise working on ARM architecture (thanks to Golang). We are redesigning the front-end, and are looking to first adapt the UI (web) component of Kerberos Open source (PHP/Laravel -> React and Golang), and make it compatible with the current Kerberos Enterprise API (which is written in Golang). Configuration changed from xml changed to json.

This mean that the back-end (machinery) will stay the same for the time being, and later on will be swapped with a modified version of Kerberos Enterprise  (add support fort USB camera + RPI camera). More updates coming..

cedricve, 2020-12-27:
This is real. Activity is logged here, announcement will come at the first of January: https://github.com/kerberos-io/opensource.

closed 2020-12-27 by cedricve