kerberos-io/machinery #201: libav-tools is not available

AnthonyKNorman, 2019-09-26:
**Describe the bug**
libav-tools is not available

**To Reproduce**
Steps to reproduce the behavior:

1. https://doc.kerberos.io/2.0/installation/Raspbian calls for 
```
sudo apt-get update && sudo apt-get install libav-tools
```

2. However, the result is
```
Package libav-tools is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  ffmpeg

E: Package 'libav-tools' has no installation candidate
```

**Expected behavior**
I expect libav-tools to install

**Screenshots**

**Operating System and Device (please complete the following information):**
 - Host system: Raspberry Pi Zero
 - OS:Raspbian Buster Lite
 - Installation Method: Generic Installation

**Additional context**

tzven0, 2019-09-26:
Hey, i think this is a raspbian buster issue. Try installing ffmpeg.

cedricve, 2019-12-03:
indeed buster has some bugs, might make sense too install ffmpeg.