kerberos-io/machinery #205: machinery v2.8.0 does not start on raspberian buster

dkasper26, 2020-01-02:
I followed http://doc.kerberos.io/opensource/installation#raspbian
Web is up, but Machinery does not start:
# /usr/bin/kerberosio
/usr/bin/kerberosio: error while loading shared libraries: libx264.so.148: cannot open shared object file: No such file or directory
# ls -l  /usr/lib/arm-linux-gnueabihf/libx264.so*
-rw-r--r-- 1 root root 1255900 Oct  3  2018 /usr/lib/arm-linux-gnueabihf/libx264.so.155
wget http://ftp.hetzner.de/ubuntu/security/pool/universe/x/x264/libx264-148_0.148.2643%2Bgit5c65704-1_armhf.deb
dpkg -i /home/pi/libx264-148_0.148.2643+git5c65704-1_armhf.deb
# ls -l  /usr/lib/arm-linux-gnueabihf/libx264.so*
-rw-r--r-- 1 root root  760820 Jan 18  2016 /usr/lib/arm-linux-gnueabihf/libx264.so.148
-rw-r--r-- 1 root root 1255900 Oct  3  2018 /usr/lib/arm-linux-gnueabihf/libx264.so.155
# /usr/bin/kerberosio
Illegal instruction
# file /usr/lib/arm-linux-gnueabihf/libx264.so*
/usr/lib/arm-linux-gnueabihf/libx264.so.148: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=779626012b9e1910c6e20ba292dde6668f2d4952, stripped
/usr/lib/arm-linux-gnueabihf/libx264.so.155: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=4d0e7af2088a20e558737748f39999f64af657e1, stripped

dkasper26, 2020-01-02:
# Workaround
I looked at the kios from the ISO:
# ls -ltr $(find /run/media | grep libx264)
lrwxrwxrwx 1 root root      14 Dec 31 10:21 /run/media/ksp/64dace40-1a90-4921-8212-0be7d45b0a46/usr/lib/libx264.so -> libx264.so.148
-rwxr-xr-x 1 root root 1199940 Dec 31 16:18 /run/media/ksp/64dace40-1a90-4921-8212-0be7d45b0a46/usr/lib/libx264.so.157
-rwxr-xr-x 1 root root  613744 Dec 31 16:18 /run/media/ksp/64dace40-1a90-4921-8212-0be7d45b0a46/usr/lib/libx264.so.148
... and copied the libx264.so.148 onto  the raspberian OS under 
/usr/lib/arm-linux-gnueabihf/libx264.so.148
... and now the Machinery is running   :-)

cedricve, 2020-01-02:
thanks currently I'm updating the documentation, and adding the shared library to the releases as well. Thanks!

cedricve, 2020-01-02:
@dkasper26 can you check the documentation again, and give it a second try? We now refer to the x264 file on the machinery repo.

Let me know if it works..

dkasper26, 2020-01-02:
Thank you very much for your quick response!
I can confirm the updated documentation with libx264-148 for rpi1 (RasPi Z) and rpi3 (RasPi 3B+)

cedricve, 2020-01-02:
Superb!

Kind regards,
Verstraeten Cédric

closed 2020-01-02 by cedricve