kerberos-io/machinery #209: Feature Request: Trigger recording from incoming MQTT message

drthanwho, 2020-03-24:
**Describe the solution you'd like**

Since there is now MQTT support for when motion is triggered, perhaps the opposite direction could be handy. For example having Machinery listen to incoming MQTT messages from motion (or door) sensors that will trigger recording (help alleviate CPU as well). So choosing to only record when receiving a MQTT message.

**Describe alternatives you've considered**

Or perhaps a way to send a message (if directly listening to mqtt topics is more complicated), to trigger recording. For example, for those using home automation platforms (HassIO), would be relatively easy to listen for motion or doors opening or anything they could think of and then send a message to kerberos to record though HassIO. This could be a MQTT message or perhaps a webhook or any other way you could think off. HassIO should have multiple way of incorporating something like that.

Stubbs, 2020-05-24:
A good use case for this is in a garden, with lots of moving plants and bushes makes it hard to define a good hull or rectangle and tweek the hueristic & algorithm to produce acurate detections. A simple PIR outside solves the problem and would be simple to make an MQTT or API request to trigger recording.

drthanwho, 2020-05-25:
Yeah I'd love to see something along those lines implemented. Even for less busy environments if I could base movement on main door opening then I'd save on computing power or even if you could combine it.
 
e.g. record only when both motion from sensor and the algorithm is detected.

neutralvibes, 2020-09-07:
Using the api you can  toggle the detectors on or off. I am using this to provide basic Alexa control.

A PUT request is what you need to toggle on and off.

### A example using curl
`
# Turning on
curl --basic -u username:password -X PUT -d active=true http://yourcamerahost/api/v1/condition/enabled
`
The data is sent in json format {"active":"false","delay":"5000"}

If no data is sent the status is returned in json format.

You can use your Home automation system etc to receive the MQTT message, then toggle the camera appropriately.