kerberos-io/machinery #210: Get Assertion desc failed at src/libswscale/swscale_internal.h:668

njjry, 2020-04-27:
**Describe the bug**
I was running kerberosio by using an IPCamera (I used raspberry pi and a rpi camera to make an IPCamera), and after it started the io devices, it gives the error: Assertion desc failed at src/libswscale/swscale_internal.h:668. I found the error was related to FFmpeg, does anyone else meets with this problem?

**To Reproduce**
Steps to reproduce the behavior:
1. make an IPCam using a raspberry pi and a rpi camera
2. input the URL in the config.xml
3. Run the kerberosio executable
4. See error

I also tried changing to VideoCapture input by feeding an mjpeg video, but it still prompt the same error. I installed the Kerberos machinery following the installation guide and used FFmpeg of version 3.1.

**Expected behavior**
Run kerberosio successfully

**Operating System and Device (please complete the following information):**
 - Host system: Linux
 - OS:  Ubuntu, 18.04
 - Installation Method: Generic Installation

**Additional context**
How to solve this problem?

cedricve, 2020-04-27:
Can you print out the ffmpeg dependencies?

    ffmpeg -v

njjry, 2020-04-27:
Thanks for replying. Below is a screenshot, 
[image]

cedricve, 2020-04-27:
It looks like you have version 3.4 installed. Please try 3.1.

njjry, 2020-04-27:
I was using 3.1, I found online saying installing FFmpeg using apt-get will solve the problem so I tried and it is 3.4 now. When it was 3.1, it also has the problem. I will show you the screenshot after I change back to 3.1.

njjry, 2020-04-27:
This is 3.1 version screenshot 
[image]
[image]
I rebuilt machinary and still the same problem.

cedricve, 2020-04-27:
ffmpeg version n3.1.11-10-g24b4c4c5ba Copyright (c) 2000-2017 the FFmpeg developers
    built with gcc 6.3.0 (Debian 6.3.0-18+deb9u1) 20170516
    configuration: --enable-nonfree --enable-libx264 --enable-gpl
    libavutil      55. 28.100 / 55. 28.100
    libavcodec     57. 48.101 / 57. 48.101
    libavformat    57. 41.100 / 57. 41.100
    libavdevice    57.  0.101 / 57.  0.101
    libavfilter     6. 47.100 /  6. 47.100
    libswscale      4.  1.100 /  4.  1.100
    libswresample   2.  1.100 /  2.  1.100
    libpostproc    54.  0.100 / 54.  0.100

I used this version which works perfectly. I see a different in the libswscale version.

njjry, 2020-04-27:
I will try, thanks

closed 2020-04-27 by njjry

cedricve, 2020-04-27:
Did it work?

njjry, 2020-04-27:
Yeah, it works. Thanks.

cedricve, 2020-04-27:
Can you explain what you did, might be interesting for other people. Thanks @njjry

njjry, 2020-04-27:
I tried to remove all these libraries using apt-get autoremove --purge, and remove the libraries under /usr/local/lib and /usr/lib, and did make uninstall. After that start from scratch to rebuild the ffmpeg.