kerberos-io/machinery #215: Use of local S3 compatible seervice

Jacq, 2020-10-05:
**Is your feature request related to a problem? Please describe.**
Could not find information about the possibility to config the S3 url:port.

**Describe the solution you'd like**
To add in the configuration the host:port info for a local S3 endpoint

**Describe alternatives you've considered**
Use S3 minio as a local storage, also  Ceph s3 should be possible if the interface expected from kerberos-io is a standard S3 endpoint.

**Additional context**
N/A.

Istar-Eldritch, 2020-10-26:
After taking a look at the source code for the cloud connection implementation it seems this is currently not supported because of the way the cloud logic manages the input for the bucket value. The current implementation generates the connection URL using string concatenation in a couple lines [here](https://github.com/kerberos-io/machinery/blob/master/src/kerberos/cloud/S3.cpp#L192) and [here](https://github.com/kerberos-io/machinery/blob/master/src/kerberos/cloud/S3.cpp#L320)

I think a good way to go about it allowing custom connection strings while keeping backwards compatibility is to just check if the connection string starts with `https` and leave it as it is if that is the case, otherwise generate the correct URL format assuming the string is the bucket name.

I'm happy to open a PR implementing the previous solution as this is something I would use myself. Happy to hear from the maintainers about what are their thoughts about it.

cedricve, 2020-11-17:
Hey @Istar-Eldritch correct, we are currently moving the entire project to a new architecture, and we will make this more configurable.