kerberos-io/machinery #217: Glitches on video capture trigger motion detection

john-bonachon, 2020-11-09:
**Describe the bug**
My camera is a TP-Link NC220. I've configured rstp for it:
```
rtsp://admin:<base64-pass>@192.168.0.20:554/h264_vga.sdp
```
and works. The issue is that it glitches every ~5-10 seconds, maybe because of frames dropped I suppose. These are some errors that ffplay report with that URL if it helps,
```
[h264 @ 0x7fd704242840] error while decoding MB 36 1, bytestream -29
[h264 @ 0x7fd704242840] concealing 1173 DC, 1173 AC, 1173 MV errors in P frame
[h264 @ 0x7fd70405c200] concealing 1131 DC, 1131 AC, 1131 MV errors in P frame
[h264 @ 0x7fd70405c200] top block unavailable for requested intra mode -1
[h264 @ 0x7fd70405c200] error while decoding MB 38 0, bytestream 473
[h264 @ 0x7fd70405c200] concealing 1200 DC, 1200 AC, 1200 MV errors in P frame
[h264 @ 0x7fd704057500] error while decoding MB 26 3, bytestream -8 
[h264 @ 0x7fd704057500] concealing 1103 DC, 1103 AC, 1103 MV errors in P frame
[rtsp @ 0x7fd704000bc0] max delay reached. need to consume packet   
[rtsp @ 0x7fd704000bc0] RTP: missed 1 packets
[rtsp @ 0x7fd704000bc0] max delay reached. need to consume packet
[rtsp @ 0x7fd704000bc0] RTP: missed 1 packets
```
They are connected via wifi router, 1 meters from each other.

These events trigger motion detection everytime, even there's no real movement. I've tried playing with the thresholds and minimum frames in sequence heuristic, but it's a big trade off sometimes.

**Expected behavior**
I expect frame dropped not be included in motion detection.

**Screenshots**

[image]

**Operating System and Device (please complete the following information):**
 - Host system: Linux VM, Raspberry Pi Version: 2
 - OS: Raspbian
 - Installation Method: Docker

**Additional context**
The ffplay output correspond to the gif taken.

cedricve, 2020-11-17:
hey there, this is typical behaviour for WIFI ip cameras. Due to frame loss (and how H264 encoding is working). If one of the frames is lost (or not properly sequences), this might show coding errors in your video. Best thing is to use IP cameras connected to Ethernet, but I understand this is not always possible. Do you have an option in the camera configuration to enable/force TCP?

john-bonachon, 2020-11-19:
I'll take a look at streams found in this [repo](https://github.com/reald/nc220). Might be something to be with UDP, and I'll need a way to find a stream by TCP. TP-Link app is smooth (no glitches) and I think it even pass through a tunnel between their servers.